OE4T/meta-tegra #2: Error trying to build core-image-sato.

pauldotknopf, 2016-05-16:
I am trying to build an example image, ```core-image-sato```.

When trying to build, I get this.

```
pknopf@ubuntu:~/git/tegra-test2/build$ bitbake core-image-sato
Parsing recipes: 100% |######################################################################################################################################################################| Time: 00:00:47
Parsing of 902 .bb files complete (0 cached, 902 parsed). 2242 targets, 230 skipped, 0 masked, 0 errors.
NOTE: There are 3 recipes to be removed from sysroot jetson-tx1, removing...
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'virtual/libgl' (but /home/pknopf/git/tegra-test2/poky/meta/recipes-graphics/libsdl/libsdl_1.2.15.bb DEPENDS on or otherwise requires it)
ERROR: tegra-libraries PROVIDES virtual/libgl but was skipped: incompatible with host aarch64-poky-linux (not in COMPATIBLE_HOST)
ERROR: lib32-mesa-gl PROVIDES virtual/libgl but was skipped: incompatible with machine jetson-tx1 (not in COMPATIBLE_MACHINE)
ERROR: mesa-gl PROVIDES virtual/libgl but was skipped: incompatible with machine jetson-tx1 (not in COMPATIBLE_MACHINE)
NOTE: Runtime target 'libsdl' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libsdl', 'virtual/libgl']
NOTE: Runtime target 'packagegroup-core-x11-sato-games' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-sato-games', 'libsdl', 'virtual/libgl']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-core-x11-sato-games', 'libsdl', 'virtual/libgl']
```

It seems that the NVidia libraries provided are only 32bit. Is this correct? What does this layer even support multi-lib then? Why not just compile everything (even then kernel) at 32bit?

pauldotknopf, 2016-05-17:
I was able to get it working using ```bitbake lib32-core-image-sato```. However, I am getting some xorg driver errors. Something to do with NVidia not supporting XORG eabi version 20, whatever that means.

pauldotknopf, 2016-05-17:
Any update on supporting 23.2?

madisongh, 2016-06-24:
I figured out how to work around the xorg ABI problem after I updated to the 24.1 drop - [https://github.com/madisongh/meta-tegra/commit/718286340bd4609a527f386accb95c8fb6f200e1](url)

This may work with the 23.2 base as well.

And yes, you have to use the lib32- prefix to get the 32-bit binaries.  For my use, I was mixing 32-bit and 64-bit userland binaries, hence the multilib configuration.

And sorry for not responding to these sooner - turns out I wasn't getting notifications about issues being opened, so I didn't know these were here.

madisongh, 2016-06-24:
At this point, the L4T R24.1/JetPack 2.2 drop has been integrated, which is fully 64-bit.  I'm not planning on supporting 32-bit kernel or userland binaries going forward.

closed 2016-06-24 by madisongh