OE4T/meta-tegra #4: Building with CUDA libs fails

lrvk123, 2016-07-31:
The deps are in place for the application that is being built:
RDEPENDS_${PN} = "cuda-toolkit cuda-cudart-dev" 
DEPENDS = "cuda-cudart"
| CMake Error at /home/ubuntu/poky/build/tmp/sysroots/x86_64-linux/usr/share/cmake-3.4/Modules/FindCUDA.cmake:617 (message):
|   Specify CUDA_TOOLKIT_ROOT_DIR
| Call Stack (most recent call first):
|   CMakeLists.txt:33 (find_package)

madisongh, 2016-07-31:
I've only tried integrating a couple of packages that use the CUDA libraries into my builds, so I don't know if this will work generally, but try adding these lines to your recipe:
```
export CUDA_BIN_PATH = "${STAGING_DIR_NATIVE}/usr/local/cuda-7.0"
EXTRA_OECMAKE += "-DCUDA_TOOLKIT_ROOT_DIR=${STAGING_DIR_TARGET}/usr/local/cuda-7.0"
LDFLAGS += "-Wl,-rpath-link,${STAGING_DIR_TARGET}/usr/local/cuda-7.0/lib -Wl,-rpath,/usr/local/cuda-7.0/lib"
```
This has worked for me, at least.  You should include `cuda-toolkit` and `cuda-toolkit-native` in your DEPENDS setting, too.

lrvk123, 2016-07-31:
Thank you, this was the solution. Now yocto picks up CUDA installation path. Previously I tried CUDA_TOOLKIT_ROOT_DIR as recommended by CMake, that did not work.

closed 2016-07-31 by madisongh