OE4T/meta-tegra #5: cuda-toolkit-7.0-76-r0 do_package QA issue: package contains non-symlink .so

lrvk123, 2016-07-31:
ERROR: cuda-toolkit-7.0-76-r0 do_package_qa: QA Issue: -dev package contains non-symlink .so: cuda-toolkit-dev path 'work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/packages-split/cuda-toolkit-dev/usr/local/cuda-7.0/lib/stubs/libnppc.so'
-dev package contains non-symlink .so: cuda-toolkit-dev path 'work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/packages-split/cuda-toolkit-dev/usr/local/cuda-7.0/lib/stubs/libcufft.so'
-dev package contains non-symlink .so: cuda-toolkit-dev path 'work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/packages-split/cuda-toolkit-dev/usr/local/cuda-7.0/lib/stubs/libcufftw.so'
-dev package contains non-symlink .so: cuda-toolkit-dev path 'work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/packages-split/cuda-toolkit-dev/usr/local/cuda-7.0/lib/stubs/libcusparse.so'
-dev package contains non-symlink .so: cuda-toolkit-dev path 'work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/packages-split/cuda-toolkit-dev/usr/local/cuda-7.0/lib/stubs/libnpps.so'
-dev package contains non-symlink .so: cuda-toolkit-dev path 'work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/packages-split/cuda-toolkit-dev/usr/local/cuda-7.0/lib/stubs/libcusolver.so'
-dev package contains non-symlink .so: cuda-toolkit-dev path 'work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/packages-split/cuda-toolkit-dev/usr/local/cuda-7.0/lib/stubs/libcublas.so'
-dev package contains non-symlink .so: cuda-toolkit-dev path 'work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/packages-split/cuda-toolkit-dev/usr/local/cuda-7.0/lib/stubs/libcurand.so'
-dev package contains non-symlink .so: cuda-toolkit-dev path 'work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/packages-split/cuda-toolkit-dev/usr/local/cuda-7.0/lib/stubs/libnppi.so' [dev-elf]
ERROR: cuda-toolkit-7.0-76-r0 do_package_qa: QA Issue: cuda-toolkit rdepends on cuda-toolkit-dev [dev-deps]
ERROR: cuda-toolkit-7.0-76-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: cuda-toolkit-7.0-76-r0 do_package_qa: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /home/ubuntu/Build/embedded_distribution/poky/build/tmp/work/aarch64-poky-linux/cuda-toolkit/7.0-76-r0/temp/log.do_package_qa.23811
ERROR: Task 208 (/home/ubuntu/Build/embedded_distribution/poky/meta-tegra/recipes-devtools/cuda/cuda-toolkit_7.0-76.bb, do_package_qa) failed with exit code '1'

lrvk123, 2016-07-31:
INSANE_SKIP_${PN}-dev += "dev-elf" seems to solve the first error.

madisongh, 2016-07-31:
Adding "dev-deps" in addition to "dev-elf" should resolve the other one.  Adding "dev-elf" should be safe, since the stub libraries are just that - stubs that are used to resolve symbols names at link time which at runtime get resolved by machine-specific libraries from the cudart package.  I believe the "dev-deps" is also a false positive for the same reason.

The way NVIDIA structured its SDK just runs at odds with what OE/Yocto expects.  I'm not sure if there's a better way to deal with it other than to disable the QA checks.  And you must be using a more restrictive set of packaging QA options than I am, since I don't think I've been seeing those errors.  I may be getting them as warnings instead.

closed 2016-08-02 by madisongh

madisongh, 2016-08-02:
I think I know why I haven't run into this before - the packages I've built that use CUDA haven't needed cuda-toolkit installed for runtime support - just cudart.  I've only needed the toolkit at build time.

commit 84c2ccda (2016-08-02): cuda-toolkit: add INSANE_SKIP flags

to clear up some package_qa errors/warnings.  Fixes #5.
commit f097308b (2016-08-02): cuda-toolkit: add INSANE_SKIP flags

to clear up some package_qa errors/warnings.  Fixes #5.