OE4T/meta-tegra #6: gcc standard library not found by nvcc

lrvk123, 2016-08-07:
Hi Matt,
I stumbled upon another error while compiling with CUDA nvcc:
standard library <new> and <cmath> are not found, below is just the first error about new: file not found. If I fix <new> manually I get next error about <cmath>.  

How do you set up gcc when compiling with nvcc for targets?

Errors below
=================================================================
_| /home/ubuntu/Build/embedded_distribution/poky/build/tmp/sysroots/x86_64-linux/usr/local/cuda-7.0/bin/nvcc -M -D__CUDACC__ /home/ubuntu/
Build/embedded_distribution/poky/build/tmp/work/aarch64-poky-linux/sensorpro/0.0.1-r0/git/core/CudaWrapper.cu -o /home/ubuntu/Build/embed
ded_distribution/poky/build/tmp/work/aarch64-poky-linux/sensorpro/0.0.1-r0/build/core/CMakeFiles/sensorprocore.dir//sensorprocore_generat
ed_CudaWrapper.cu.o.NVCC-depend -ccbin /home/ubuntu/Build/embedded_distribution/poky/build/tmp/sysroots/x86_64-linux/usr/bin/aarch64-poky
-linux/aarch64-poky-linux-gcc -m64 -arch=sm_20 -std=c++11 -O2 -DVERBOSE -g -DNVCC -I/home/ubuntu/Build/embedded_distribution/poky/build/t
mp/sysroots/jetson-tx1/usr/local/cuda-7.0/include -I/home/ubuntu/Build/embedded_distribution/poky/build/tmp/sysroots/jetson-tx1/usr/local
/cuda-7.0/samples/common/inc -I/home/ubuntu/Build/embedded_distribution/poky/build/tmp/sysroots/jetson-tx1/usr/include -I/home/ubuntu/Bui
ld/embedded_distribution/poky/build/tmp/work/aarch64-poky-linux/sensorpro/0.0.1-r0/build -I/home/ubuntu/Build/embedded_distribution/poky/
build/tmp/work/aarch64-poky-linux/sensorpro/0.0.1-r0/git -I/home/ubuntu/Build/embedded_distribution/poky/build/tmp/sysroots/jetson-tx1/us
r/local/cuda-7.0/include
| In file included from /home/ubuntu/Build/embedded_distribution/poky/build/tmp/sysroots/jetson-tx1/usr/local/cuda-7.0/include/cuda_runti
me.h:105:0,
|                  from <command-line>:0:
| /home/ubuntu/Build/embedded_distribution/poky/build/tmp/sysroots/jetson-tx1/usr/local/cuda-7.0/include/common_functions.h:87:15: fatal 
error: new: No such file or directory
| compilation terminated.
| CMake Error at sensorprocore_generated_CudaWrapper.cu.o.cmake:203 (message):
|   Error generating
|   /home/ubuntu/Build/embedded_distribution/poky/build/tmp/work/aarch64-poky-linux/sensorpro/0.0.1-r0/build/core/CMakeFiles/sensorprocor
e.dir//./sensorprocore_generated_CudaWrapper.cu.o
| 
|_

madisongh, 2016-08-08:
I don't think gcc will find the C++ standard headers by default unless you tell it you're compiling C++ code.  Can you add `-xc++` to CPPFLAGS?

lrvk123, 2016-08-08:
Thank you for your advice. I started by rebuilding toolchain, downgrading gcc-5 to gcc-4 as this might be an issue according to CUDA documentation. If this fails, I continue by adding C++ flags as you recommended. I will report back here when I find anything conclusive.

madisongh, 2016-09-20:
Still having this problem?

madisongh, 2016-10-12:
Closing this as there has been no follow up for a while.

closed 2016-10-12 by madisongh