OE4T/meta-tegra #9: getrandom and nvidia 3.10 kernel

jackmitch, 2016-09-12:
I'm hitting an issue compiling gnutls with what I assume is the tx1 kernel. Are you having the sames issues? You must be using an oe-core with gnutls 3.5.3 or greater. The gnutls feature that broke it I believe to be:

** libgnutls: Use getrandom where available via the syscall interface.
   This works around an issue of not-using getrandom even if it exists
   since glibc doesn't declare such function.

As far as I understand, the nvidia 3.10 kernel doesn't implement it, and the fallback is also failing?

madisongh, 2016-09-12:
I haven't run into that issue with gnutls specifically, but I have hit it with some other packages - although for those, the fallback (to using /dev/random) seems to work OK.  I've still patched those packages to skip trying the getrandom() call, just to cut down on console noise from the syscall dispatcher.

For my production work, I'm on the krogoth branch.  It pulls in gnutls 3.4.9, which may explain why I'm not seeing this particular issue.

madisongh, 2016-09-20:
What version of linux-libc-headers are you using?  I used to set PREFERRED_VERSION_linux-libc-headers to "3.16" in jetson-tx1.conf, but removed that with [this commit](https://github.com/madisongh/meta-tegra/commit/f85e0aa4c817444d448918c1eb87e891add215fd) - it wasn't really appropriate for machine configuration, and while some missing syscalls get logged at runtime (getrandom being a typical culprit), packages I'm running appear to work OK in spite of it.

If I put back the setting, I see gnutls fail to build.  So I think this one is already resolved by that change - let me know if you're still seeing it.

jackmitch, 2016-09-20:
This fixes the  build for me to, thanks.

closed 2016-09-20 by jackmitch