OE4T/meta-tegra #12: Flash nvidia jetson tx1

GuzMp, 2016-09-29:
Hi i'm new in this project, so i don't understand all..
First, before to put our yocto layer, we must to download Jetson TX1 64-bit Driver Package or all is integrated in this meta-tegra ?
Second, how we put this in the nvidia jetson tx1 ?
And third, how we choice our target for the bitbake (core-image-minimal, core-image-sato, ...) ?

Thank you for your attention.

GuzMp, 2016-09-30:
I don't how we build all recipes in a .tar.
I must to build each recipe, one by one ? Or it's possible to build all in one bitbake command ?

madisongh, 2016-09-30:
The recipes in the layer download the needed binary and source packages.  You don't need to install JetPack or L4T on your build host in advance.

For getting it onto the Jetson TX1 development kit, there are two approaches.  The first is to put the built rootfs image on an SDcard.  This is pretty simple, and the Jetson will automatically boot from the SDcard if it's inserted in the slot on the board.  It also lets you keep the stock Ubuntu image on the on-board eMMC so you can easily switch back to that when needed.

The second approach is to flash the built rootfs onto the eMMC.  To do that, you set IMAGE_FSTYPE to "tegraflash" in your image recipe, then follow the instructions on [this wiki page](https://github.com/madisongh/meta-tegra/wiki/Flashing-the-Jetson-TX1-without-being-root).

As for which image recipe to choose, it's up to you.  core-image-minimal is a good choice to begin with, but core-image-sato should build and boot as well.

madisongh, 2016-09-30:
> I don't how we build all recipes in a .tar.
I must to build each recipe, one by one ? Or it's possible to build all in one bitbake command ?

I'm not exactly sure what you're asking here.  If you're referring to a tarball that contains sources for multiple packages, there is no fixed answer.  In the meta-tegra recipes, I've taken a couple of different approaches for this, since NVIDIA does this sort of bundling with L4T.

GuzMp, 2016-09-30:
Thank you very much! But I've a point that I don't understand.
When you talk about  "IMAGE_FSTYPE to "tegraflash" in your image recipe", what and where is image recipe ?

GuzMp, 2016-09-30:
So if I choose core-image-sato, I set IMAGE_FSTYPE to "tegraflash" in core-image-sato ?

madisongh, 2016-09-30:
> So if I choose core-image-sato, I set IMAGE_FSTYPE to "tegraflash" in core-image-sato ?

In that recipe, or through a .bbappend for that recipe, yes.  If you're not sure what an image recipe is, you might want to familiarize yourself with the [Yocto Project documentation](https://www.yoctoproject.org/documentation) before getting too deep into this.

GuzMp, 2016-09-30:
Thank you, so to recap in core-image-sato.bb there is :

`DESCRIPTION = "Image with Sato, a mobile environment and visual style for \
mobile devices. The image supports X11 with a Sato theme, Pimlico \
applications, and contains terminal, editor, and file manager."

IMAGE_FEATURES += "splash package-management x11-base x11-sato ssh-server-dropbear hwcodecs"

LICENSE = "MIT"

inherit core-image

IMAGE_INSTALL += "packagegroup-core-x11-sato-games"

IMAGE_FSTYPE = "tegraflash"`

madisongh, 2016-09-30:
Yes, that should do it.

GuzMp, 2016-10-03:
Once I've build with "bitbake core-image-sato", I've untar "core-image-sato-jetson-tx1.tar.gz" located in "build/tmp/deploy/images/jetson-tx1" but I don't find "doflash.sh".
I've just bin, boot, dev, etc, home, lib, media, mnt, proc, run, sbin, sys, tmp, usr, var.
I've just found flash.sh in /build/tmp/work-shared/tegra-binaries-24.2.0-r0/Linux_for_Tegra, so it's the NVIDIA L4T package. 
I think I missed a point..

madisongh, 2016-10-03:
Assuming you're building off master, there should also be a `core-image-sato-jetson-tx1.tegraflash.zip` in `tmp/deploy/images/jetson-tx1`.  That's the file that contains the rootfs, bootloader images, and `doflash.sh` script.

madisongh, 2016-10-03:
Except I just noticed a typo in your prior message that I missed earlier.  The variable to set in the image recipe is IMAGE_FSTYPES, not IMAGE_FSTYPE.

GuzMp, 2016-10-03:
Oh yes! And now when I building I've 5 errors because of IMAGE_FSTYPES 
ERROR: /poky/meta/recipes-sato/images/core-image-sato-sdk.bb: No IMAGE_CMD defined for IMAGE_FSTYPES entry 'tegraflash' - possibly invalid type name or missing support class
ERROR: /poky/meta/recipes-sato/images/core-image-sato.bb: No IMAGE_CMD defined for IMAGE_FSTYPES entry 'tegraflash' - possibly invalid type name or missing support class
ERROR: /poky/meta/recipes-sato/images/core-image-sato-sdk-ptest.bb: No IMAGE_CMD defined for IMAGE_FSTYPES entry 'tegraflash' - possibly invalid type name or missing support class
ERROR: /poky/meta/recipes-sato/images/core-image-sato-dev.bb: No IMAGE_CMD defined for IMAGE_FSTYPES entry 'tegraflash' - possibly invalid type name or missing support class
ERROR: Failed to parse recipe: /poky/meta/recipes-sato/images/core-image-sato-sdk-ptest.bb

madisongh, 2016-10-03:
Ah, right, oops... I haven't yet added this to the machine configuration, so please try adding the following line to the image recipe as well: `IMAGE_CLASSES += "image_types_tegra"`

GuzMp, 2016-10-03:
I've added IMAGE_CLASSES += "image_types_tegra" after IMAGE_FSTYPES = "tegraflash" but I've still the same errors..

madisongh, 2016-10-03:
Can you try either moving that line to above the `inherit core-image` line in the recipe, or adding it instead to your `$BUILDDIR/conf/local.conf` file?

GuzMp, 2016-10-03:
When I move above the inherit core-image, I've now this errors :
ERROR: Nothing PROVIDES '${IMAGE_UBOOT}'
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', '${IMAGE_UBOOT}']

I put just IMAGE_CLASSES += "image_types_tegra" in local.conf or IMAGE_CLASSES and IMAGE_FSTYPES ? 
When I put just IMAGE_CLASSES or IMAGE_CLASSES and IMAGE_FSTYPES in local.conf, there is the same error : 
ERROR: Nothing PROVIDES '${IMAGE_UBOOT}'
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', '${IMAGE_UBOOT}']

GuzMp, 2016-10-03:
I've updated your layer, and it's good, I haven't error. I will see tomorrow if I've good files !
Thank you very much for you help!

GuzMp, 2016-10-04:
Now when the building is finished I have new error : 

ERROR: core-image-sato-1.0-r0 do_image_tegraflash: Function failed: do_image_tegraflash (log file is located at /home/GuzMp/poky/build_jetson/tmp/work/jetson_tx1-poky-linux/core-image-sato/1.0-r0/temp/log.do_image_tegraflash.13083)
ERROR: Logfile of failure stored in: /home/GuzMp/poky/build_jetson/tmp/work/jetson_tx1-poky-linux/core-image-sato/1.0-r0/temp/log.do_image_tegraflash.13083
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: Python function set_image_size finished
| DEBUG: Executing shell function do_image_tegraflash
| open input file /core-image-sato-jetson-tx1.ext3 failed.
| WARNING: exit code 255 from a shell command.
| ERROR: Function failed: do_image_tegraflash (log file is located at /home/GuzMp/poky/build_jetson/tmp/work/jetson_tx1-poky-linux/core-image-sato/1.0-r0/temp/log.do_image_tegraflash.13083)
ERROR: Task 16 (/home/GuzMp/poky/meta/recipes-sato/images/core-image-sato.bb, do_image_tegraflash) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5390 tasks of which 15 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  /home/GuzMp/poky/meta/recipes-sato/images/core-image-sato.bb, do_image_tegraflash
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

madisongh, 2016-10-04:
It looks from the error message like you might be mixing poky's krogoth branch with meta-tegra's master branch.   meta-tegra's branch names correspond with poky's (or oe-core's), and mixing them will likely result in errors.  If you are looking to build an image using the L4T R24.2 content, and you're using poky krogoth, use the krogoth-l4t-r24.2 branch in meta-tegra instead.

GuzMp, 2016-10-04:
Sorry but I don't understand your last line, I don't know how to do this.

GuzMp, 2016-10-05:
What is krogoth-l4t-r24.2 branch ?

madisongh, 2016-10-05:
It is a branch in this git repository.

GuzMp, 2016-10-05:
It's possible to explain more what I have to do please.. 
Scuse me for the disturbance, I want to learn but I'm French I've some difficult in English.

GuzMp, 2016-10-05:
I don't find the krogoth-l4t-r24.2 branch

GuzMp, 2016-10-06:
I don't touch DISTRO ?= "poky" in build_jetson/conf/local.conf ?
krogoth-l4t-r24.2 is a layer like meta-tegra ? Where it is ?

GuzMp, 2016-10-06:
I've finish the build without error, I've put IMAGE_CLASSES += "image_types_tegra" and IMAGE_FSTYPES = "tegraflash" at the end of build/conf/local.conf.
I've unzip core-image-sato-jetson-tx1.tegraflash.zip and inside there are : 
Archive:  ../core-image-sato-jetson-tx1-20161006094916.tegraflash.zip
  inflating: nvtboot_cpu.bin         
  inflating: nvtboot_recovery.bin    
  inflating: nvtboot.bin             
  inflating: gpt.img                 
  inflating: cboot.bin               
  inflating: tegra210-jetson-tx1-p2597-2180-a01-devkit.dtb  
  inflating: ppt.img                 
  inflating: doflash.sh              
  inflating: core-image-sato.img     
  inflating: jetson-tx1.cfg          
  inflating: flash.xml               
  inflating: bpmp.bin                
  inflating: warmboot.bin            
  inflating: tos.img                 
  inflating: board_config_jetson-tx1.xml  

I should not have others files ?

madisongh, 2016-10-06:
Glad you got the build to finish.  I don't see `u-boot-jetson-tx1.bin` in your list.  That file should be present in the zip archive, too.

madisongh, 2016-10-06:
As far as your questions about branching go... how did you create your build workspace?  Did you not use a `git clone` command to download the metadata layers you are using for your build?  Are you not familiar with how branching works in a version control system like git?

GuzMp, 2016-10-06:
I explain step by step what I do in command-line : 

git clone -b krogoth git://git.yoctoproject.org/poky.git 
cd poky
git clone https://github.com/madisongh/meta-tegra
. oe-init-build-env build_jetson
cd build_jetson/conf/

**Then I add meta-tegra in bblayers.conf**
nano bblayers.conf
BBLAYERS ?= " \
  /home/teosit/poky/meta \
  /home/teosit/poky/meta-poky \
  /home/teosit/poky/meta-yocto-bsp \
  /home/teosit/poky/meta-tegra \
  "
**Then I SET MACHINE in local.conf**
nano local.conf
MACHINE ??= "jetson-tx1"

**And put at the end**
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"

**Then I build**
cd ..
bitbake core-image-sato

**Once the building finished, I go unzip core-image-sato-jetson-tx1.tegraflash.zip**
cd tmp/deploy/images/jetson-tx1
mkdir image_jetson
sudo unzip core-image-sato-jetson-tx1.tegraflash.zip

And I've just the files that I've put above. I've missed a step ?

GuzMp, 2016-10-06:
I see u-boot-jetson-tx1.bin it's not in core-image-sato-jetson-tx1.tegraflash.zip
It's all files in poky/build_jetson/tmp/deploy/images/jetson-tx1 : 

core-image-sato-jetson-tx1-20161006094916.rootfs.ext3
core-image-sato-jetson-tx1-20161006094916.rootfs.manifest
core-image-sato-jetson-tx1-20161006094916.tegraflash.zip
core-image-sato-jetson-tx1.ext3
core-image-sato-jetson-tx1.manifest
core-image-sato-jetson-tx1.tegraflash.zip
Image
Image--3.10+git0+271164dd77-r0-jetson-tx1-20161006094916.bin
Image--3.10+git0+271164dd77-r0-tegra210-jetson-tx1-p2597-2180-a01-devkit-20161006094916.dtb
image_jetson
Image-jetson-tx1.bin
Image-tegra210-jetson-tx1-p2597-2180-a01-devkit.dtb
modules--3.10+git0+271164dd77-r0-jetson-tx1-20161006094916.tgz
modules-jetson-tx1.tgz
README_-_DO_NOT_DELETE_FILES_IN_THIS_DIRECTORY.txt
u-boot-dtb.bin
u-boot-jetson-tx1-2015.07+gitAUTOINC+1d3dd062ca-r0.bin
u-boot-jetson-tx1.bin

And in the directory _image_jetson_ there is core-image-sato-jetson-tx1.tegraflash.zip unzip

GuzMp, 2016-10-06:
There should be no a directory rootfs, bootloader and kernel ?

madisongh, 2016-10-06:
> git clone -b krogoth git://git.yoctoproject.org/poky.git 
The `-b krogoth` specifies the branch in the poky repository that you cloned.  To be compatible, you must also check out either the `krogoth` branch (which is based off L4T R24.1) or the `krogoth-l4t-r24.2` branch (for L4T R24.2) in the meta-tegra repository.  That will ensure that the recipes and configuration files in the meta-tegra layer are compatible with those under poky/meta.

So to fix this:
```
cd /home/teosit/poky/meta-tegra
git checkout krogoth-l4t-r24.2
```
Then re-run your build.

madisongh, 2016-10-06:
> There should be no a directory rootfs, bootloader and kernel ?

The `core-image-sato-jetson-tx1.img` file holds the rootfs and kernel, formatted for programming into the APP partition in the flash.  The bootloader file that should be there (`u-boot-jetson-tx1.bin` mentioned earlier) is missing.

GuzMp, 2016-10-06:
Ah yes thank you, I understand better now ! And when the build is completed, I have to add tegraflash.py in the directory and execute in command-line "sudo ./doflash.sh" or "sudo ./doflash.sh jetson-tx1 mmcblk0p1" ? (with jetson-tx1 in recovery mode)

madisongh, 2016-10-06:
> And when the build is completed, I have to add tegraflash.py in the directory

Unfortunately, tegraflash.py needs other files and programs, so you can't just copy that into the directory.  This should work:
```
sudo PATH=$BUILDDIR/tmp/sysroots/x86_64-linux/usr/bin:$PATH ./doflash
```

GuzMp, 2016-10-07:
Oh.. I've re-run after do this : 
cd /home/teosit/poky/meta-tegra
git checkout krogoth-l4t-r24.2

And return another error : 
Image Name:   U-Boot 2015.07-rc2 for p2371-218
| Created:      Fri Oct  7 11:20:09 2016
| Image Type:   ARM U-Boot Firmware (uncompressed)
| Data Size:    441048 Bytes = 430.71 kB = 0.42 MB
| Load Address: 80110000
| Entry Point:  80110000
| Image Name:   U-Boot 2015.07-rc2 for p2371-218
| Created:      Fri Oct  7 11:20:09 2016
| Image Type:   ARM U-Boot Firmware (uncompressed)
| Data Size:    453949 Bytes = 443.31 kB = 0.43 MB
| Load Address: 80110000
| Entry Point:  80110000
| make[1]: Leaving directory '/home/teosit/poky/build/tmp/work/jetson_tx1-poky-linux/u-boot-tegra/2015.07+gitAUTOINC+1d3dd062ca-r0/build'
| make: Leaving directory '/home/teosit/poky/build/tmp/work/jetson_tx1-poky-linux/u-boot-tegra/2015.07+gitAUTOINC+1d3dd062ca-r0/git'
| Traceback (most recent call last):
|   File "/home/teosit/poky/build/tmp/sysroots/x86_64-linux/usr/bin/elf-get-entry.py", line 38, in <module>
|     f = file(sys.argv[1], 'rb')
| IOError: [Errno 2] No such file or directory: '/home/teosit/poky/build/tmp/work/jetson_tx1-poky-linux/u-boot-tegra/2015.07+gitAUTOINC+1d3dd062ca-r0/git//u-boot'
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/teosit/poky/build/tmp/work/jetson_tx1-poky-linux/u-boot-tegra/2015.07+gitAUTOINC+1d3dd062ca-r0/temp/log.do_compile.15674)
ERROR: Task (/home/teosit/poky/meta-tegra/recipes-bsp/u-boot/u-boot-tegra_2015.07.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4474 tasks of which 11 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/teosit/poky/meta-tegra/recipes-bsp/u-boot/u-boot-tegra_2015.07.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

madisongh, 2016-10-07:
Try `bitbake -c clean u-boot-tegra` then `bitbake -c compile u-boot-tegra`.  I suspect the error is due to your prior build with the mixed master/krogoth setup.

GuzMp, 2016-10-07:
No because I had remove and build a new. 
I try these command but there is the same error :

rectoyon, 2016-10-08:
@madisongh do you need to do anything after ./doflash.sh? I'm getting this at the end of my dmesg
```
failed to init firmware from filesystem: tegra21x_xusb_firmware
```
so Ethernet won't come up.

madisongh, 2016-10-08:
Do you see that firmware under /lib/firmware in your rootfs?  It should be getting pulled in as part of the tegra-firmware package.  If it's there, then either the kernel firmware loader should find it, or the udev firmware helper should.

rectoyon, 2016-10-08:
yep, it's in /lib/firmware.

From what I've read https://devtalk.nvidia.com/default/topic/968645/jetson-tx1/24-2-fails-to-initialise-usb/2 it seems that the kernel is trying to load the firmware before the filesystem that has /lib/firmware is mounted.

I see in the dmesg a bunch of USB stuff happening around 2.7 and then the filesystem mount around 6.8.

But, I also read here https://devtalk.nvidia.com/default/topic/928383/tegra21x_xusb_firmware-failing-to-load-/ about some udev versions not being compatible.

One thing I did notice was that the line 
```
<filename> u-boot-jetson-tx1. </filename>
```
in flash.xml was missing the .bin. Would this be related at all? Does just doing ./doflash.sh do everything I need?

madisongh, 2016-10-08:
The missing suffix in flash.xml wouldn't cause this, but I'll look into it, since it should be there.

You're right about some udev versions not being compatible, though - I remember before I had switched over to using systemd that I had to patch back in the firmware helper support into the version of eudev that was in oe-core at the time, since it had been removed as no longer necessary.  systemd's udev still supports it, so it hasn't bothered me for a while (which is why I hadn't remembered initially).

rectoyon, 2016-10-08:
i'm not sure if this is another clue or not, but sato just throws a splash screen and then goes to a black screen/prompt and doesn't do anything else after that. I've only been able to get images such as core-image-minimal to get me to a prompt.

rectoyon, 2016-10-09:
@madisongh so this worked

https://devtalk.nvidia.com/default/topic/787495/jetson-tk1/-solved-l4t-21-1-enable-usb-3-0-/2

I noticed that the default value for CONFIG_EXTRA_FIRMWARE had a bunch of NVIDIA firmware files for some sort of Android tablet. Do you think this is an acceptable solution?

madisongh, 2016-10-09:
I've seen that approach mentioned on the linux-tegra mailing list for the upstream kernel, too.  It would certainly help, especially for booting from a USB rootfs.

The approach NV took with L4T was to use an initrd.  That's viable, too, I guess, although I'd like to give folks the option of using whichever mechanism works best for their particular application, and not carry the extra baggage of either the firmware-in-kernel or building an initrd if they don't actually need it.

GuzMp, 2016-10-09:
Madisongh now when I build there is new error : 
ERROR: Nothing RPROVIDES 'gstreamer1.0-omx-tegra' (but /home/p-a/poky/meta/recipes-sato/images/core-image-sato.bb RDEPENDS on or otherwise requires it)
ERROR: gstreamer1.0-omx-tegra was skipped: because it has a restricted license not whitelisted in LICENSE_FLAGS_WHITELIST
NOTE: Runtime target 'gstreamer1.0-omx-tegra' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['gstreamer1.0-omx-tegra']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'gstreamer1.0-omx-tegra']

madisongh, 2016-10-09:
I'll fix that shortly.  For now, add
```
LICENSE_FLAGS_WHITELIST_append = " commercial_gstreamer1.0-omx-tegra"
```
to your local.conf file.

madisongh, 2016-10-09:
@rectoyon 
> I'm not sure if this is another clue or not, but sato just throws a splash screen and then goes to a black screen/prompt and doesn't do anything else after that.

I saw that once in my testing this morning; restarting the X server `/etc/init.d/xserver-nodm restart` cleared it up.  I couldn't reproduce it, though - on each reboot, the Sato UI came up after the splash screen just fine.  Some kind of race condition, perhaps, or maybe some difference between my display (a generic small HDMI monitor) and yours?

GuzMp, 2016-10-10:
Once jetson-tx1 flashed, then I start and at the end : 
[    8.085704] udevd[732]: did not find firmware file 'gm20b/gpmu_ucode_desc.bin'
[    8.094583] udevd[736]: did not find firmware file 'gm20b/pmu_sig.bin'
[    8.102306] udevd[736]: did not find firmware file 'gm20b/fecs_sig.bin'
[    8.110392] udevd[736]: did not find firmware file 'gm20b/acr_ucode.bin'
[    8.118607] udevd[736]: did not find firmware file 'gm20b/pmu_bl.bin'
[    8.139101] udevd[736]: did not find firmware file 'gm20b/gpu2cde.bin'
[    8.142115] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.156593] udevd[736]: did not find firmware file 'gm20b/gpu2cde.bin'
[    8.164883] udevd[736]: did not find firmware file 'gm20b/gpu2cde.bin'
[    8.173082] udevd[736]: did not find firmware file 'gm20b/gpu2cde.bin'
(EE)
Fatal server error:
(EE) no screens found(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

GuzMp, 2016-10-10:
Then I click "enter" and ask login so I enter "root" and once logged, there are no directory or files. It's' normal ?

madisongh, 2016-10-10:
I'm not sure if those messages from udevd are actual errors (the firmware may still be getting found in its actual location), but if you could `ln -sn tegra21x /lib/firmware/gm20b` to  get rid of them.

As for the X server errors - do you have a monitor connected to the HDMI port, one that worked with the NVIDIA Ubuntu distro?  I get that error if the X server starts but my monitor is disconnected or powered off.

> Then I click "enter" and ask login so I enter "root" and once logged, there are no directory or files. It's' normal ?

From just an `ls` command, yes.  `/home/root` has no contents.  You should find files elsewhere in the filesystem.

GuzMp, 2016-10-10:
I don't use HDMI port, I using serial port

madisongh, 2016-10-10:
No problem, then.  You can ignore the X server errors (and those firmware errors, too).

GuzMp, 2016-10-11:
Thank you, it works!
I summarize for help other people : 

git clone -b krogoth git://git.yoctoproject.org/poky.git
cd poky

git clone https://github.com/madisongh/meta-tegra
cd meta-tegra
git checkout krogoth-l4t-r24.2

. oe-init-build-env build_jetson
cd build_jetson/conf/

**Then you add meta-tegra in bblayers.conf**
nano bblayers.conf
BBLAYERS ?= " \
/home/teosit/poky/meta \
/home/teosit/poky/meta-poky \
/home/teosit/poky/meta-yocto-bsp \
/home/teosit/poky/meta-tegra \
"
**Then you SET MACHINE in local.conf**
nano local.conf
MACHINE ??= "jetson-tx1"
**And put at the end**
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
LICENSE_FLAGS_WHITELIST_append = " commercial_gstreamer1.0-omx-tegra"

**Then you build**
cd ..
bitbake core-image-sato

**Once the building finished, you go unzip core-image-sato-jetson-tx1.tegraflash.zip**
cd tmp/deploy/images/jetson-tx1
sudo unzip core-image-sato-jetson-tx1.tegraflash.zip
**replace [y]es** 

**Then you have to add the following directory in order to add tools flash:**
sudo PATH=$BUILDDIR/tmp/sysroots/x86_64-linux/usr/bin:$PATH ./doflash.sh

**Once flash finished, you can control your jetson-tx1 with Serial Port in opening a screen :**
sudo screen /dev/ttyUSB0 115200

madisongh, 2016-10-12:
Great to hear, and thanks for the summary.  I'll close this out now.

closed 2016-10-12 by madisongh