OE4T/meta-tegra #13: do_image_tegraflash fails

rectoyon, 2016-10-08:
Working through https://github.com/madisongh/meta-tegra/issues/12

and having 
```
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
```
at the end of my local.conf.

 I get this when I try to bitbake core-image-sato (core-image-minimal works fine)
```
| DEBUG: Executing python function set_image_size
| DEBUG: Python function set_image_size finished
| DEBUG: Executing shell function do_image_tegraflash
| size of /opt/Yocto/krogoth/tmp/deploy/images/jetson-tx1/core-image-sato-jetson-tx1.ext3 is not mulple of 4096.
| WARNING: exit code 255 from a shell command.
| ERROR: Function failed: do_image_tegraflash
```
I tried turning on debugging and looking at the log file but there's not much information beyond the above. be curious if you have any thoughts on things I may be doing wrong.

rectoyon, 2016-10-08:
I can't seem to find the source to the underlying function mksparse so not really sure why this works. But, playing around with the block size this seems to work
```
mksparse -b 1024 -v --fillpattern=0 "${DEPLOY_DIR_IMAGE}/${IMAGE_LINK_NAME}.ext3" ${IMAGE_BASENAME}.img
```

madisongh, 2016-10-08:
The mksparse binary is part of the L4T package, and NVIDIA unfortunately doesn't provide the source for it. 

Another approach would be to add
```
IMAGE_ROOTFS_ALIGNMENT = "4096"
```
to your local.conf.

I'm just curious as to why I'm not seeing this issue.

rectoyon, 2016-10-08:
you mean 
```
IMAGE_ROOTFS_ALIGNMENT = "1024"
```
yeah?

Yes, not sure. You provided a great suggestion so I'll just go with that.

closed 2016-10-08 by rectoyon