OE4T/meta-tegra #14: tegra210_xusb_firmware needs to be available during early boot

madisongh, 2016-10-09:
As mentioned in #12, the xusb firmware needs to be available during early kernel initialization, particularly if the rootfs is on a USB device.

Possible approaches to resolve this:

1. Build the firmware into the kernel image via CONFIG_EXTRA_FIRMWARE.  This might not be great from a licensing perspective.
1. For non-USB boots, use systemd-udevd instead of eudev.  systemd-udevd includes the firmware loading helper, which was removed from eudev.  Patches for eudev to restore that functionality could also be added.
1. Use an initrd that at least has the needed firmware.  This is the approach taken by L4T.

madisongh, 2016-10-09:
Using a minimal initrd with the firmware in it solves the USB problem, but appears to cause problems with later firmware loads (after the real rootfs has been mounted).  So I suspect we'd have to go all-in with a more-complete initrd for that approach.  Otherwise, one still needs a udev firmware helper.

For now, I'll add in eudev patches to make the firmware helper work, then revisit after some further investigation.

rectoyon, 2016-10-10:
@madisongh I did a clean build last night and tried this out this morning and eth0 comes up fine. Thanks for the patch.

rectoyon, 2016-10-11:
@madisongh noticing today that I need to restart networking after boot in order to get eth0 up
```
/etc/init.d/networking restart
```

madisongh, 2016-10-11:
Any errors in the boot log about it?   Race between the USB coming up and the networking initscript, perhaps?

rectoyon, 2016-10-11:
I see the firmware get loaded
```
root@jetson-tx1:~# dmesg | grep firmware
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    7.691201] udevd[177]: writing '/lib/firmware/tegra21x_xusb_firmware' (124416) to '/sys/devices/platform/tegra-xhci/firmware/tegra21x_xusb_firmware/data'
[    7.712337] tegra-xhci tegra-xhci: Firmware File: tegra21x_xusb_firmware (124416 Bytes)
```
It looks to me like the networking initscript is being called too early?
```
 6.352108] ipip: IPv4 over IPv4 tunneling driver
[    6.357232] ip_tables: (C) 2000-2006 Netfilter Core Team
[    6.362706] arp_tables: (C) 2002 David S. Miller
[    6.367352] TCP: cubic registered
[    6.370663] Initializing XFRM netlink socket
[    6.375075] NET: Registered protocol family 10
[    6.380359] mip6: Mobile IPv6
[    6.383377] ip6_tables: (C) 2000-2006 Netfilter Core Team
...
```

Maybe massage the order of the init scripts?

rectoyon, 2016-10-13:
As another clue it seems that on a fresh install eth0 always comes up. But, upon a reboot eth0 is missing. I'm trying to think of persistent elements that would cause this behavior. I tried disabling the udev-cache but that didn't do the trick.

madisongh, 2016-10-14:
Hmm.  I haven't seen this on reboots in my testing with the sato image, but there could be a race where eth0 isn't showing up in time for the networking initscript to see it.  Which image are you running?

rectoyon, 2016-10-14:
it's my own based on core-image-minimal with a bit more debugging/network stuff in there so I bet its some issue with something delaying things in an out of order way. When I get a chance I'll try a more standard image.

madisongh, 2016-11-11:
I have seen this issue when it takes a long time for the USB controller to probe all of the attached USB peripherals - in my case, this was a custom board with a custom USB device attached that the controller could see but could not complete all of the protocol handshaking.  I expect that using some udev rules to trigger bringing the interface up when it becomes visible would help in this situation.

madisongh, 2016-12-06:
Having the initrd in place to load the xusb controller's firmware during early boot (which is also done in the stock L4T Ubuntu image) takes care of this for typical use cases.

closed 2016-12-06 by madisongh