OE4T/meta-tegra #15: package cuda-binaries fails

rectoyon, 2016-10-12:
@madisongh I'm not sure this is really an issue or not. I'm trying to get access to the cuda tools, such as cuda-gdbserver.

I've added "cuda-binaries" to my IMAGE_INSTALL packages but I get this 
```
ERROR: Task do_populate_sdk in /opt/oe/krogoth/meta-tempmine/recipes-core/image/core-image-mimbal.bb rdepends upon non-existent task do_populate_sysroot in /opt/oe/krogoth/meta-tegra/recipes-devtools/cuda/cuda-binaries_8.0.34-1.bb
ERROR: Command execution failed: 1
```

I see a lot of cuda-* packages. Maybe I'm including the wrong one? what's a good set to include?

madisongh, 2016-10-12:
Try "cuda-toolkit" instead.  "cuda-binaries" is just used to download the .deb repository; other recipes depend on it so they can all share the same download (which is really large).

rectoyon, 2016-10-12:
ok, have that. thanks.

But, I don't see cuda-gdbserver on the target or things such as nvcc in my resultant sdk. Any thoughts on what i'm doing wrong?

rectoyon, 2016-10-12:
From the cuda-toolkit recipe I see this get expanded out
```
cuda-gdb-src-8-0_8.0.34-1_arm64.deb
```
is there a way to build it?

madisongh, 2016-10-12:
Ah, right.  There are so many packages there, and I guess I didn't get enough of them to fully flesh out on-target development.  You probably need to add cuda-command-line-tools and cuda-core, at least, to the CUDA_PKGS definition in the recipe to get the additional programs.

rectoyon, 2016-10-12:
hmm, tried adding command-line-tools, core, and toolkit but no joy. The gdb one in there is source code so it feels like it needs to be explicitly built. Should I use --extract and then is there a command to build for that .deb file?

rectoyon, 2016-10-12:
ok, it's starting to sorta come together for me. I added these lines to the recipe. 
```
install -d ${D}${prefix}/local/cuda-8.0/bin
cp -R --preserve=mode,timestamps ${B}/usr/local/cuda-8.0/bin/* ${D}${prefix}/local/cuda-8.0/bin
FILES_${PN} = "${prefix}/local/cuda-8.0/lib/*${SOLIBS} ${prefix}/local/cuda-8.0/lib/stubs ${prefix}/local/cuda-8.0/bin/*"
```
I now see cuda-gdbserver in the resultant filesystem.

The native/host stuff is confusing to me. Is there a way to build the tools for both the target and the host? e.g., so they show up when I do populate_sdk?

madisongh, 2016-10-12:
It's confusing to me, too, which is why the recipes aren't really complete.  I haven't even looked into how to fit them into a Yocto-style SDK yet.  I've only put enough pieces in place to cross-build some CUDA apps within an OE/Yocto build workspace.  I'm hoping to get more pieces together over time, but it's complicated and will take time to understand how they all fit together.

This is one of the downsides of trying to wedge these binary-only packages into something other than the specific distro that NVIDIA builds them for.

rectoyon, 2016-10-12:
yeah, that makes sense. I'll take it as a work in progress. I'm hesitant to fork it as I don't feel good about my own recipe but fwiw the lines above will move the toolkit binaries into the target filesystem. Feel free to close this case if you feel that's appropriate.

madisongh, 2016-10-14:
I've updated the cuda-toolkit recipe to bring in the command line tools, similar to what you mentioned above, but I've put the .../bin/* files into the -dev package, rather than the main package.  I think that's more correct, but as you say, this is a work in progress....

closed 2016-10-14 by madisongh

rectoyon, 2016-10-15:
thanks Matt. I'll give it a whirl tomorrow.

rectoyon, 2016-10-16:
@madisongh looking good. Thanks for addressing this.