OE4T/meta-tegra #16: add ROS in the project

GuzMp, 2016-10-13:
Hi! Once I built my img with core-image-sato, I want to add ROS in jetson-tx1.
So for that, I created an initial build directory with in my local.conf meta-ros and his dependencies. 
Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "aarch64-poky-linux"
MACHINE           = "jetson-tx1"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1.1"
TUNE_FEATURES     = "aarch64"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "krogoth:204b2bae4a5958735fdd19c63a69f4ed3780bba7"
meta-oe           
meta-python       
meta-multimedia   = "krogoth:851a064b53dca3b14dd33eaaaca9573b1a36bf0e"
meta-tegra        = "krogoth-l4t-r24.2:c5494deee41d2ec1165e2490339117b7f771a731"
meta-ros          = "master:35cb79a0a1cd4ec6e92a197ec1d4f5d26ea8ae29"

Then I run with core-image-ros-roscore. 
But the building completed, after I don't know how to add this in my project with core-image-sato img in order to flash all in jetson-tx1.

Thank you.

madisongh, 2016-10-13:
The core-image-ros-roscore should build a complete image, and if you still have the IMAGE_CLASSES and IMAGE_FSTYPES settings in your local.conf file, your build should have generated a core-image-ros-roscore-jetson-tx1.tegraflash.zip file for it.  Just flash that image to the eMMC.

If you added the IMAGE_xxx settings to the core-image-sato.bb recipe rather than your local.conf file, just make the same changes to core-image-ros-roscore.bb to generate the zip file.

GuzMp, 2016-10-14:
Ok thank you but it's not possible instead of build a complete image, just build an image that I could add with core-image-sato and flash both ?

madisongh, 2016-10-14:
If I understand you correctly, you're asking how to add packages from the meta-ros layer into your existing core-image-sato build, rather than building a completely separate image.

You can do this by listing the packages you want to add in the CORE_IMAGE_EXTRA_INSTALL setting in your local.conf file.  There are other ways to customize your image, too -- see [this section](http://www.yoctoproject.org/docs/2.1/mega-manual/mega-manual.html#usingpoky-extend-customimage) of the Yocto Project documentation.

GuzMp, 2016-10-14:
Sorry it's contrary, I have my core-image-sato and I don't want to modify. Now I want to build a image for ROS. So I want to build completely separate image and at the end flash both in the jetson-tx1.

madisongh, 2016-10-14:
The easiest way to switch between two images on the Jetson is to put one of them on an SDcard - with an SDcard inserted, the Jetson will boot from there instead of the on-board flash.  Just pop out the SDcard to switch back again.  I don't currently include anything in the layer to generate an SDcard image automatically, but it's not too hard to format an SDcard and copy an ext3 or ext4 rootfs image generated by your build onto it.

A more complicated method is to modify the partitioning of the on-board flash so you have two partitions for root filesystems, then switch between them by marking one or the other as bootable.

madisongh, 2016-12-06:
Haven't heard anything more on this, so I'll assume any issues have been resolved.

closed 2016-12-06 by madisongh