OE4T/meta-tegra #18: nvcc does not find cicc

sbaccam90, 2016-10-25:
Hi,

I compiled core-image-sato-sdk system, it boots and everything works except that when I try to compile in the Jetson board using nvcc it throws an error:

sh: cicc: command not found

After reading in the blog of NVIDIA, I understood that the error is produced because nvcc is not finding nvvm (cicc). I reviewed the nvcc.profile file an effectively, the route of the file does not exists in the root file system. The weirdest part is that in the Packages.gz file of cuda-repo-l4t-8-0-local_8.0.34-1_arm64.deb, cicc is not mentioned. In equivalent files, such cuda-repo-ubuntu1504-7-5-local_7.5-18_amd64.deb, the Packages.gz files indicates that cicc is part of the cuda-core-<ver>.deb package.  I checked that core, command-line-tools, and other packages appear in the list of cuda-toolkit_8.0.34-1.bb file. Also, I added  IMAGE_INSTALL+= " cudnn cuda-toolkit" to local.conf file. What do you think can be the problem?

Thank you very much, you did an amazing job developing meta-tegra.

Sergio

sbaccam90, 2016-10-26:
I solved my problem adding the folder nvvm manually after flashing the internal SD. I think it can be done inside the cuda-toolkit<ver>.bb file, because this directory is not copied in the do_install task. However, I tried to add this to the task and I obtained  some errors because apparently I am trying to override a folder that cuda-cudart<ver>.bb already created.

madisongh, 2016-10-26:
Yes, the /usr/local/cuda-8.0/nvvm subtree wasn't being repackaged with the rest of the toolkit, an oversight on my part.  I've updated the recipe to bring it in.

closed 2016-10-26 by madisongh