OE4T/meta-tegra #20: Unable to use omxh265enc in gstreamer

stammw, 2016-11-23:
I'm trying to have this command to work:

```
gst-launch-1.0 fakesrc \
    ! omxh265enc ! 'video/x-h265, stream-format=(string)byte-stream' \
    ! filesink location=test.mkv -
```

Everything is OK running this on Nvidia's Ubuntu system, but on the one I built with oe, it gives:

```
WARNING: erroneous pipeline: no element "omxh265enc"
```

gst-inspect-1.0 output doesn't show all the `omx*` pipelines :

```
root@jetson-tx1:~# gst-inspect-1.0 |grep omx
omx:  omxmpeg4videodec: OpenMAX MPEG4 Video Decoder
omx:  omxh264dec: OpenMAX H.264 Video Decoder
omx:  omxmpeg4videoenc: OpenMAX MPEG4 Video Encoder
omx:  omxaacenc: OpenMAX AAC Audio Encoder
```

Wheras all of these are present in nvidia the system:

```
ubuntu@tegra-ubuntu:~$ gst-inspect-1.0  | grep omx
omx:  omxmpeg4videodec: OpenMAX MPEG4 Video Decoder
omx:  omxh264dec: OpenMAX H.264 Video Decoder
omx:  omxh265dec: OpenMAX H.265 Video Decoder
omx:  omxvp8dec: OpenMAX VP8 Video Decoder
omx:  omxvp9dec: OpenMAX VP9 Video Decoder
omx:  omxmpeg2videodec: OpenMAX MPEG2 Video Decoder
omx:  omxwmvdec: OpenMAX WMV Video Decoder
omx:  omxaacdec: OpenMax AAC decoder
omx:  omxmpegaudiodec: OpenMax MP3/MP2 decoder
omx:  omxh264enc: OpenMAX H.264 Video Encoder
omx:  omxh265enc: OpenMAX H.265 Video Encoder
omx:  omxvp8enc: OpenMAX VP8 Video Encoder
omx:  nvhdmioverlaysink: OpenMax HDMI Video Sink
omx:  nvoverlaysink: OpenMax Video Sinkk
```

Some omxh265enc seem to be compiled into libgstomx.so though.

```
root@jetson-tx1:~# grep -c omxh265enc /usr/lib/gstreamer-1.0/libgstomx.so
3
```

Am I missing some configuration options or depency ?

Here is my OE local.conf : 

```
MACHINE ?= "jetson-tx1"
DISTRO_FEATURES = "opengl x11"
IMAGE_INSTALL_append = " \
kernel \
kernel-devicetree \
kernel-image \
kernel-modules \
kernel-vmlinux \
libdrm \
gstreamer1.0-libav \
gstreamer1.0-plugins-bad \
gstreamer1.0-plugins-base \
gstreamer1.0-plugins-tegra \
gstreamer1.0-plugins-good \
gstreamer1.0-omx-tegra \
gstreamer1.0-plugins-nveglgles"
```

And the configuration information :

```
Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "aarch64-oe-linux"
MACHINE           = "jetson-tx1"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "aarch64"
TARGET_FPU        = ""
meta              = "krogoth:28da89a20b70f2bf0c85da6e8af5d94a3b7d76c9"
meta-tegra        = "krogoth-l4t-r24.2.1:119e0fd8c2f49b33fbd5a5849f3926a8e3c58eb2"
```

stammw, 2016-11-23:
I just found out that the /etc/xdg/gstomx.conf file lacks some configuration I found in `/blah/oe-jetson-tx1/build/tmp-glibc/work/aarch64_tegra210-oe-linux/gstreamer1.0-omx-tegra/1.0.0-r24.2.0-r0/gstomx1_src/gst-omx1/gstomx_config.c`, for exemple : 

```
[omxh265enc]
type-name=GstOMXH265Enc
core-name=/usr/lib/libnvomx.so
component-name=OMX.Nvidia.h265.encoder
rank=266
```

Shouldn't this file be correctly set out of the box ?

madisongh, 2016-11-23:
I don't have that configuration file in my builds, so the compiled-in configuration gets used and the plugin works for h264 and h265.    Do you know where that file came from?  What happens if you remove it?

stammw, 2016-11-23:
I've finally been able to encode a h265 video with hardware support (thank you). But I don't know where this configuration file is from, and I can't check it right now but I think it might be from a DISTRO_FEATURES = "x11" ?

stammw, 2016-11-24:
It seems that this file is installed by `tmp-glibc/work/aarch64_tegra210-oe-linux/gstreamer1.0-omx-tegra/1.0.0-r24.2.0-r0/gstomx1_src/gst-omx1/config/rpi/Makefile.am` :

```
if USE_OMX_TARGET_RPI
configdir = $(sysconfdir)/xdg
config_DATA = gstomx.conf
endif
```

I don't know what could have defined `USE_OMX_TARGET_RPI`. I tried to activate / desactivate `DISTRO_FEATURE = "x11"` but this config file never came back from a fresh build...

closed 2016-12-06 by madisongh

rectoyon, 2017-07-19:
@stammw I'm having issues on the TX2 #37. Any chance you've tried omx out on the TX2?

stammw, 2017-07-19:
I haven't tried it yet.

sushitek, 2023-03-15:
> 

Can you share how you managed it?