OE4T/meta-tegra #26: core-image-minimal: Bootloader fails to perform autoboot from eMMC on Jetson TX1

selbstereg, 2017-04-07:
Hello madisongh,

first of all thank you for this BSP. We have the following problem:

We basically followed [issue #12](https://github.com/madisongh/meta-tegra/issues/12) and were
successful with the core-image-sato, but with core-image-minimal the boot process failed.

The flashing looks good but when U-Boot scans mmc 0:1, it doesn't recognize anything to boot from. Indeed `ls mmc 0:1 boot` reveals, that /boot is empty.

We tried this with branch morty-l4t-r27.1 as well as with commit 1436d4c2ad8dd1ec3d96f475629a4d7b1c354116 from master.

Do you have any clues for us what to try next?

Thanks in advance,

selbstereg

madisongh, 2017-04-09:
Yes, I ran into this as well.  I may need to move kernel-image and kernel-device tree from MACHINE_EXTRA_RRECOMMENDS to MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS, since the minimal rootfs only includes the latter, and for booting any of the Tegra systems we pretty much need those files all the time.

In the meantime, you can work around it by adding
```
IMAGE_INSTALL_append = " ${MACHINE_EXTRA_RRECOMMENDS}"
```
to your local.conf.

selbstereg, 2017-04-10:
Awesome, that worked. Thank you very much and have a good time maintaining your great layer.

closed 2017-04-10 by selbstereg