OE4T/meta-tegra #29: tegraflash on TX2

rectoyon, 2017-05-12:
@madisongh Any thoughts on the error below. 

When I add 
```
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
```
to the end of my local.conf  and try to bitbake core-image-sato I get
```
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'u-boot'
ERROR: u-boot-tegra PROVIDES u-boot but was skipped: incompatible with machine jetson-tx2 (not in COMPATIBLE_MACHINE)
ERROR: u-boot was skipped: PREFERRED_PROVIDER_virtual/bootloader set to u-boot-tegra, not u-boot
ERROR: u-boot-tegra PROVIDES u-boot but was skipped: incompatible with machine jetson-tx2 (not in COMPATIBLE_MACHINE)
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'u-boot']
```
without adding the IMAGE_FSTYPES it builds fine.

madisongh, 2017-05-12:
I haven't yet worked out all the necessary pieces to make a tegraflash-able image for the TX2, so you can only build an SDcard image at this point.

Besides U-Boot, it looks like the tegraflash tooling and configuration is a bit different for TX2 than for TX1, so the bbclass will need updating, too.  I've just been too busy with my day job lately to get to it.

rectoyon, 2017-05-12:
got it. Thanks for the quick reply.

JackKiefer, 2017-06-15:
Hey there!
Awesome work you're doing. 
I would also very much like to be able to get a tegraflash-able TX2 image. Your stuff is working awesome for the TX1, by the way. Thanks for your time and all you've done!

madisongh, 2017-06-25:
I am working on supporting flashing the TX2, but the differences between TX1 and TX2 are significant, so it will take some time.  I may hold off on doing this until R28 is released.

madisongh, 2017-07-15:
I have pushed tegraflash support for the TX2 to master for experimentation and testing.

closed 2017-07-15 by madisongh

rectoyon, 2017-07-17:
@madisongh thanks for the updates! Got sato working. I'll keep plowing ahead trying a few different things.

rectoyon, 2017-07-17:
@madisongh so the plus side is the flashing works great.

not sure I want to open a new issue for this. But, having some issues writing new images/dtb to the /boot directory. If I put a new file there I get this
```
ext4fs_devread read outside partition 12002609648
```
and then it bails on retrieving the kernel

I've had issues with ext4 and u-boot in the past. I was wondering if you've seen anything like this.

My immediate thought would be to make /boot a separate partition and make it ext2 or vfat.

rectoyon, 2017-07-18:
@madisongh I changed ext3 to ext4 in image_types_tegra.bbclass and it seems to be working now.

madisongh, 2017-07-18:
OK, I'll make the switch.   This could the what's causing #36, too.

rectoyon, 2017-07-18:
hmm, hadn't seen #36. well that's interesting, I see
```
/dev/mmcblk0p1  1.2G  893M  157M  86% /
```
I'll play with it and see if I can come up with anything.