OE4T/meta-tegra #30: CMake FindCuda.cmake bails out

graugans, 2017-05-24:
I am using this layer together with meta-ros. When I try to compile the ``image-geometry`` recipe I get the following error:

```console
Loading cache...done.
Loaded 3145 entries from dependency cache.
Parsing recipes...done.
Parsing of 2379 .bb files complete (2375 cached, 4 parsed). 3149 targets, 197 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.34.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "aarch64-poky-linux"
MACHINE           = "jetson-tx1"
DISTRO            = "poky"
DISTRO_VERSION    = "2.3"
TUNE_FEATURES     = "aarch64"
TARGET_FPU        = ""
meta              
meta-poky         = "HEAD:e5aa1ac23669a45491de9ecffac491c01e70203f"
meta-oe           
meta-multimedia   
meta-python       
meta-networking   = "HEAD:5e82995148a2844c6f483ae5ddd1438d87ea9fb7"
meta-tegra        = "HEAD:dd2912a420796eb4fb3742dca670721146941930"
meta-ros          = "HEAD:aebd5b3195b35a5a15c8e21397e357dae25b7fa2"

Initialising tasks...done.
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Running task 2122 of 2122 (/data/test/jetson-ros-bsp/sources/meta-ros/recipes-ros/vision-opencv/image-geometry_1.11.15.bb:do_configure)
NOTE: recipe image-geometry-1.11.15-r0: task do_configure: Started
Log data follows:
| DEBUG: Executing shell function do_configure
| NOTE: cmake.bbclass no longer uses OECMAKE_BUILDPATH.  The default behaviour is now out-of-tree builds with B=WORKDIR/build.
| -- The C compiler identification is GNU 5.4.0
| -- The CXX compiler identification is GNU 5.4.0
| -- Check for working C compiler: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
| -- Check for working C compiler: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CXX compiler: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++
| -- Check for working CXX compiler: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Using CATKIN_DEVEL_PREFIX: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/build/devel
| -- Using CMAKE_PREFIX_PATH: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot/opt/ros/indigo;/data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot/usr;/data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/opt/ros/indigo;/data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr
| -- This workspace overlays: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot/opt/ros/indigo;/data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot/usr;/data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/opt/ros/indigo;/data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr
| -- Found PythonInterp: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/bin/python-native/python (found version "2.7.13")
| -- Using PYTHON_EXECUTABLE: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/bin/python-native/python
| -- Using default Python package layout
| -- Found PY_em: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc
| -- Using empy: /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc
| -- Using CATKIN_ENABLE_TESTING: 0
| -- catkin 0.6.19
| CMake Error at /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
|   Could NOT find CUDA (missing: CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) (found
|   suitable exact version "8.0")
| Call Stack (most recent call first):
|   /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
|   /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/share/cmake-3.7/Modules/FindCUDA.cmake:1013 (find_package_handle_standard_args)
|   /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/toolchain.cmake:42 (find_package)
|   /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot/usr/share/OpenCV/OpenCVConfig.cmake:105 (find_host_package)
|   CMakeLists.txt:5 (find_package)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/temp/log.do_configure.15756)
NOTE: recipe image-geometry-1.11.15-r0: task do_configure: Failed
NOTE: Tasks Summary: Attempted 2122 tasks of which 2121 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /data/test/jetson-ros-bsp/sources/meta-ros/recipes-ros/vision-opencv/image-geometry_1.11.15.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

I also tried to inherit the cuda.bbclass in the ``image-geometry_1.11.15.bb`` recipe

```patch
diff --git a/recipes-ros/vision-opencv/image-geometry_1.11.15.bb b/recipes-ros/vision-opencv/image-geometry_1.11.15.bb
index 3ed5a84..f2cbc9a 100644
--- a/recipes-ros/vision-opencv/image-geometry_1.11.15.bb
+++ b/recipes-ros/vision-opencv/image-geometry_1.11.15.bb
@@ -6,3 +6,5 @@ LIC_FILES_CHKSUM = "file://package.xml;beginline=12;endline=12;md5=d566ef916e9de
 DEPENDS = "opencv sensor-msgs"
 
 require vision-opencv.inc
+
+inherit cuda

```

madisongh, 2017-05-25:
Adding the `inherit cuda` should have helped... can you shoot me a copy of the `run.do_configure` and the `log.do_configure` that got generated?

graugans, 2017-05-29:
Thanks @madisongh for your quick reply. I've uploaded the requested files to [a gist](https://gist.github.com/graugans/30d3eb280a5050fe6aa4032515c591d5) You may need to scroll a bit left to see the cuda CMake defines:
```
-DCUDA_TOOLKIT_TARGET_DIR=/data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot/usr/local/cuda-8.0   -DCUDA_TOOLKIT_ROOT_DIR=/data/test/jetson-ros-bsp/build/tmp/work/jetson_tx1-poky-linux/image-geometry/1.11.15-r0/recipe-sysroot-native/usr/local/cuda-8.0
```

madisongh, 2017-05-30:
The cmake invocation looks OK, as far as I can tell.  Could you try running it with cmake's tracing and debug output  enabled? That should produce a more detailed log, which hopefully will point us in the right direction.  You could either do that by editing the `run.do_configure` script to add `--trace-expand --debug-output` to the `cmake` command and running it manually, or with `EXTRA_OECMAKE += "--trace-expand --debug-ouput"` in the recipe.

graugans, 2017-05-31:
I've uploaded the requested output to the [gist](https://gist.githubusercontent.com/graugans/30d3eb280a5050fe6aa4032515c591d5/raw/f922a7fd915027efc9feb9ea0843092573f50693/cmake_trace_log.do_configure) Thanks for pointing out the cmake ``--trace-expand --debug-output``

madisongh, 2017-05-31:
This looks like further brokenness in cmake's FindCUDA module.  After the `inherit cuda`, try adding the following:
```
EXTRA_OECMAKE += "-DCUDA_TOOLKIT_ROOT_DIR_INTERNAL=${STAGING_DIR_NATIVE}/usr/local/cuda-${CUDA_VERSION}"
EXTRA_OECMAKE += "-DCUDA_TOOLKIT_TARGET_DIR_INTERNAL=${STAGING_DIR_TARGET}/usr/local/cuda-${CUDA_VERSION}"
EXTRA_OECMAKE += "-DCUDA_USE_STATIC_CUDA_RUNTIME=OFF"
```
The first two lines prevent FindCUDA from ignoring the location you already gave it and trying to locate the CUDA toolkit again by itself.  The third line works around a check for pthreads before which it completely munges CMAKE_C_FLAGS so the check can never work in the cross-build environment.

graugans, 2017-06-01:
Perfect! This solved my issue. In theory this should fail for any other user of the ``cuda.bbclass`` or am I wrong?

madisongh, 2017-06-02:
I believe it would, yes.

madisongh, 2017-06-03:
Should be fixed now, in master and pyro.

closed 2017-06-03 by madisongh