OE4T/meta-tegra #31: tegraflash no rootfs?

rectoyon, 2017-05-26:
@madisongh i'm following the instructions at the end of this issue https://github.com/madisongh/meta-tegra/issues/12

it seems to go ok but it doesn't appear to flash a kernel or rootfs. After it reboots I just get a u-boot prompt. the doflash.sh happens very quickly but does seem to end ok
```
[   3.2199 ] Flashing completed

[   3.2199 ] Coldbooting the device
[   3.2206 ] tegradevflash --reboot coldboot
[   3.2211 ] Cboot version 00.01.0000
[   3.2231 ] 
```

Is there a next step I need to do?

madisongh, 2017-05-27:
Can you try core-image-base instead, or add ${MACHINE_EXTRA_RRECOMMENDS} to your core-image-minimal images?  I think I need to move a couple of the packages listed in that variable to MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS instead, so the kernel image, device tree, and extlinux.conf file get installed in `/boot`.

You may also need to explicitly set IMAGE_ROOTFS_ALIGNMENT to "4096" as mentioned in that other thread.

closed 2017-06-03 by madisongh

paroque28, 2024-07-10:
If anyone needs this information... this could also happen if you forget to remove the following configuration:
`TNSPEC_BOOTDEV = "nvme0n1p1"`

commit 877cb466 (2017-06-03): conf/machine: fix minimal rootfs builds

- make kernel-image and kernel-devicetree essential
- set IMAGE_ROOTFS_ALIGNMENT

[ Fixes #31 ]
commit 3369bcaf (2017-06-03): conf/machine: fix minimal rootfs builds

- make kernel-image and kernel-devicetree essential
- set IMAGE_ROOTFS_ALIGNMENT

[ Fixes #31 ]