OE4T/meta-tegra #32: sdcard image for jetson-tk1

twoerner, 2017-05-30:
Building for "core-image-minimal" or "core-image-base" I don't seem to end up with anything in tmp-glibc/deploy/images/jetson-tk1 that would obviously be dd'ed to an sdcard:

core-image-base-jetson-tk1-20170530054448.rootfs.manifest
core-image-base-jetson-tk1-20170530054448.rootfs.tar.gz
core-image-base-jetson-tk1-20170530054448.testdata.json
core-image-base-jetson-tk1.manifest -> core-image-base-jetson-tk1-20170530054448.rootfs.manifest
core-image-base-jetson-tk1.tar.gz -> core-image-base-jetson-tk1-20170530054448.rootfs.tar.gz
core-image-base-jetson-tk1.testdata.json -> core-image-base-jetson-tk1-20170530054448.testdata.json
...

There is the *rootfs.tar.gz, but wouldn't that assume I've already partitioned and formatted the card?

madisongh, 2017-05-30:
Right.  The MACHINE configs don't set IMAGE_FSTYPES, so just set that to "ext3" or "ext4" in your local.conf to get a filesystem image you can dd onto a partition.

twoerner, 2017-05-31:
Wow, what a journey! I hadn't used my TK1 yet, but had bought it a while ago (years!). My firmware was *so* old... (how old was it?!) it wouldn't do the boot-from-sdcard thing. This was quite frustrating. However, after spending the last day or so learning about and flashing R21.5.0 to the eMMC, I can now report that my build is up and running on my TK1!

Thanks! :-D

- add "ext4" to IMAGE_FSTYPES
- build
- partition sdcard with one partition of type 83 (Linux)
- format this partition (# mkfs.ext4 /dev/sd?1)
- load build image to sdcard (# dd if=<image> of=/dev/sd?1 bs=10M)

OpenEmbedded nodistro.0 jetson-tk1 /dev/ttyS0

jetson-tk1 login: root
root@jetson-tk1:~# uname -a
Linux jetson-tk1 3.10.40+g71cc3bfd7f05 #1 SMP PREEMPT Mon May 29 20:33:17 EDT 2017 armv7l GNU/Linux

closed 2017-05-31 by twoerner