OE4T/meta-tegra #33: How to Enable TEGRA_DRM? ( for wayland )

JoshJYM, 2017-06-05:
Hi 

Now I am build AGL with meta-tegra layer.
So, I get AGL image for jetson-tx1 EVK  successfully Thanks.

But When I boot AGL on jetson-tx1 EVK, I can't see weston-launcher screen.
So, I excuted weston-laucher manually and get follow errors.

```
root@jetson-tx1:~# /usr/bin/weston-launch --tty /dev/tty1 --user root -- --idle-time=4294967
Date: 2017-06-02 UTC
[03:34:31.978] weston 1.11.0
               http://wayland.freedesktop.org
               Bug reports to: https://bugs.freedesktop.org/enter_bug.cgi?product=Wayland
component=weston&version=1.11.0
               Build: 1.10.93-2-g2d825ed configure.ac: bump to version 1.11.0 for the official release (2016-05-31 17:10:40 -0700)
[03:34:31.978] OS: Linux, 3.10.96+, #14 SMP PREEMPT Mon Jun 5 18:42:52 KST 2017, aarch64
[03:34:31.979] Using config file '/etc/xdg/weston/weston.ini'
[03:34:31.979] Output repaint window is 7 ms maximum.
[03:34:31.979] Loading module '/usr/lib/weston/drm-backend.so'
[03:34:31.982] initializing drm backend
[03:34:31.982] logind: failed to get session seat
[03:34:31.982] logind: cannot setup systemd-logind helper (-61), using legacy fallback
[03:34:31.983] no drm device found
[03:34:31.983] fatal: failed to create compositor backend
failed to restore keyboard mode: Bad file descriptor
failed to set KD_TEXT mode on tty: Bad file descriptor
```

I think that I need DRM for executing weston-launcher. 
So, I added TEGRA_DRM using 'make menuconfig'.
But Still I got same error as well. I think DRM is not working very well.

madisongh, 2017-06-05:
My knowledge of this is limited, but as far as I know, NVIDIA does not provide the binary bits needed to use the DRM backend for weston on the Jetson boards, so no, you can't use weston-launcher. You should be able to use XWayland - I think the NVIDIA-supplied libraries provide enough support for that.

JoshJYM, 2017-06-08:
Official kernel version for jetson-tx1 is 3.10. 
If I upgrade kernel version, Can I use DRM?

Because I saw another github which try to enable DRM.
But kernel is not official, 4.9.2  

Thank you for your response!

closed 2017-06-25 by madisongh