OE4T/meta-tegra #35: TX1: X server unsuported ABI

hesmar, 2017-06-30:
Hi,
when I try to run the X server on the TX1, I get the message that the X server video driver ABI version 23.0 is not supported by the nvidia driver. Is it possible to solve this somehow?

madisongh, 2017-06-30:
Yes, I've run into that before and did [this](https://github.com/madisongh/meta-tegra/commit/718286340bd4609a527f386accb95c8fb6f200e1#diff-f76b3105b15e49c74ebd7eff023fa83b) to work around it.  Depending on how the ABI has changed, that may not work, and the only solution would be to go back to an older version of X that's compatible with what NVIDIA provides.

hesmar, 2017-06-30:
Ok, I also tried to add the ignoreABI flag but then I only get a black screen. So it does not seem to work. That means that I have to use another Yocto release or is it possible to just downgrade the X server?

madisongh, 2017-10-24:
I have pulled in the xserver-xorg 1.18.4 recipe on the `pyro-l4t-r24.2.2` branch, which appears to work well. Look like some other layers do something similar for X server compatibility.

I expect this could be extended to other branches where there's an ABI mismatch between the X server provided by OE-core and the NVIDIA-supplied extension.

madisongh, 2018-03-25:
This is the approach I've taken when running to other cases of ABI incompatibility (e.g. TK1 with L4T R21.6).  Will use this going forward, should it crop up again.

closed 2018-03-25 by madisongh