OE4T/meta-tegra #36: Something wonky with the TX1 partition table

JackKiefer, 2017-07-10:
I've got an issue that I'm wondering might have been encountered by others before. 
After flashing the TX-1 using the `doflash.sh` script, something strange seems to have happened with the partition table.

Running `lsblk`, we get some pretty normal-looking results:

```
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0rpmb  179:48   0    4M  0 disk 
mmcblk0boot0 179:16   0    4M  1 disk 
mmcblk0boot1 179:32   0    4M  0 disk 
mmcblk0      179:0    0 14.7G  0 disk 
|-mmcblk0p1  179:1    0   12G  0 part /
|-mmcblk0p2  179:2    0    2M  0 part 
|-mmcblk0p3  179:3    0    4M  0 part 
|-mmcblk0p4  179:4    0    2M  0 part 
|-mmcblk0p5  179:5    0    6M  0 part 
|-mmcblk0p6  179:6    0    4M  0 part 
|-mmcblk0p7  179:7    0    6M  0 part 
|-mmcblk0p8  179:8    0    2M  0 part 
|-mmcblk0p9  179:9    0    2M  0 part 
|-mmcblk0p10 179:10   0   20M  0 part 
|-mmcblk0p11 179:11   0   64M  0 part 
|-mmcblk0p12 179:12   0   64M  0 part 
|-mmcblk0p13 179:13   0    4M  0 part 
|-mmcblk0p14 179:14   0    2M  0 part 
|-mmcblk0p15 179:15   0    6M  0 part 
|-mmcblk0p16 259:0    0    6M  0 part 
|-mmcblk0p17 259:1    0    2M  0 part 
`-mmcblk0p18 259:2    0  2.5G  0 part
```
But taking a look at our actually useable space with `df -ah`, something is up:

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1  667M  566M   65M  90% /
```

This is further confirmed by du:
```
root@TX1:/# du -sbh /
612M    /
```

Anyone have any idea as to what might be going on or how to fix this? Thanks!

madisongh, 2017-07-11:
Have you run fsck on the partition to see if the filesystem can be repaired?

JackKiefer, 2017-07-12:
The filesystem _can_ be repaired, but I'm running on a headless distro without parted or anything. I've worked around by simply flashing to an SD at the moment.

madisongh, 2017-07-13:
OK, that indicates that it's not the partition table, but the filesystem within the partition, that's at fault.

Could be the filesystem image that gets built is bad, although that doesn't seem likely - you could check it by runing fsck on the `.ext3` file.  It could also be the `mksparse` tool out of L4T that's introducing some corruption.

commit d5bf53f2 (2017-07-18): image_types_tegra: switch from ext3 to ext4

to correct some issues mentioned in #29, and
possibly fix #36.
commit 46376afc (2017-07-18): image_types_tegra: switch from ext3 to ext4

to correct some issues mentioned in #29, and
possibly fix #36.
closed 2017-07-18 by madisongh