OE4T/meta-tegra #54: e1000e ethernet driver not working?

sakoman, 2017-11-03:
I can see a few initial messages from the driver at boot:

[    2.530680] e1000e: Intel(R) PRO/1000 Network Driver - 3.2.6-k
[    2.530683] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
[    2.530720] igb: Intel(R) Gigabit Ethernet Network Driver - version 5.3.0-k
[    2.530722] igb: Copyright (c) 2007-2014 Intel Corporation.

But it never reaches the point where it binds to eth1.

Is there anything special I need to add to my image contents?  As far as I can tell there is no firmware file required for the e1000e.

Is anyone else seeing this issue?

madisongh, 2017-11-03:
Is this a PCI Ethernet card you have plugged in?

sakoman, 2017-11-04:
I using the Astro Carrier for Jetson TX2 from Connect Tech Inc:

http://connecttech.com/product/astro-carrier-for-nvidia-jetson-tx2-tx1/

It has two ethernet jacks, one handled by the eqos driver (eth0) and the other by the e1000e driver (eth1)

With the Nvidia Ubuntu image/kernel I see these messages:

[   27.305185] e1000e: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
<snip>
[   28.783974] eqos 2490000.ether_qos eth0: Link is Up - 1Gbps/Full - flow control rx/tx

With the Yocto image I only get:

[   36.539340] eqos 2490000.ether_qos eth0: Link is Up - 1Gbps/Full - flow control rx/tx

Digging a little deeper, I see that your kernel config and the kernel config in the Ubuntu kernel are the same (other than a handful of CAN configs).  I also checked the firmware folders in the two images and saw nothing different of any significance.

The ubuntu kernel boot messages for the e1000e driver look like this:

[    2.456985] e1000e: Intel(R) PRO/1000 Network Driver - 3.2.6-k
[    2.456987] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
[   22.000521] e1000e 0000:01:00.0: Disabling ASPM L0s L1
[   22.000548] e1000e 0000:01:00.0: enabling device (0000 -> 0002)
[   22.000863] e1000e 0000:01:00.0: Interrupt Throttling Rate (ints/sec) set to dynamic conservative mode
[   22.106448] e1000e 0000:01:00.0 eth1: registered PHC clock
[   22.106451] e1000e 0000:01:00.0 eth1: (PCI Express:2.5GT/s:Width x1) 00:0c:8b:90:06:88
[   22.106453] e1000e 0000:01:00.0 eth1: Intel(R) PRO/1000 Network Connection
[   22.106545] e1000e 0000:01:00.0 eth1: MAC: 3, PHY: 8, PBA No: FFFFFF-0FF
[   27.305185] e1000e: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None

With the Yocto image we only get the first two:

[    2.532307] e1000e: Intel(R) PRO/1000 Network Driver - 3.2.6-k
[    2.532309] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.

Any ideas on what I might be missing?

sakoman, 2017-11-04:
I tracked down the issue.

I needed to explicitly include the kernel-modules package in my image recipe, since the machine config doesn't include /lib/modules/4.4.38-l4t-r28.1+g79e4600/kernel/drivers/net/ethernet/intel/ixgbe/ixgbe.ko, which is required for the driver to function properly!

closed 2017-11-04 by sakoman