OE4T/meta-tegra #58: u-boot problems

jhaws1982, 2017-12-12:
I'm having 2 problems with the image generated with tegraflash:

1. u-boot-jetson-tx1.bin flashes properly but will not boot. I've been running morty, but have also tried pyro-l4t-r28.1 with the same results. I get the following output on the serial console from morty:

<LOTS OF OUTPUT DURING FLASH - NO ERRORS NOTED>
...
[0068.504] partition DTB write successful.
[0068.515] Writing BCT partition
[0068.518] bct_init bctinit
[0068.602] partition BCT write successful.
[0068.610] Change state to cold boot mode
[0068.614] Flushing data to stroage
[0000.183] [TegraBoot] (version 24.00.2015.42-mobile-ec3b827e)
[0000.189] Processing in cold boot mode Bootloader 2
[0000.193] A02 Bootrom Patch rev = 63
[0000.196] Power-up reason: software reset
[0000.200] No Battery Present
[0000.203] RamCode = 0
[0000.205] Platform has Ddr4 type ram
[0000.209] max77620 disabling SD1 Remote Sense
[0000.213] Setting Ddr voltage to 1125mv
[0000.217] Serial Number of Pmic Max77663: 0x811ad
[0000.224] Entering ramdump check
[0000.227] Get RamDumpCarveOut = 0x0
[0000.231] RamDumpCarveOut=0x0,  RamDumperFlag=0xe59ff3f8
[0000.236] Last reboot was clean, booting normally!
[0000.240] Sdram initialization is successful
[0000.244] SecureOs Carveout Base=0xff800000 Size=0x00800000
[0000.250] GSC1 Carveout Base=0xff700000 Size=0x00100000
[0000.255] GSC2 Carveout Base=0xff600000 Size=0x00100000
[0000.260] GSC3 Carveout Base=0xff500000 Size=0x00100000
[0000.265] GSC4 Carveout Base=0xff400000 Size=0x00100000
[0000.270] GSC5 Carveout Base=0xff300000 Size=0x00100000
[0000.276] BpmpFw Carveout Base=0xff2c0000 Size=0x00040000
[0000.281] Lp0 Carveout Base=0xff2bf000 Size=0x00001000
[0000.296] RamDump Carveout Base=0xff23f000 Size=0x00080000
[0000.302] Platform-DebugCarveout: 0
[0000.305] Nck Carveout Base=0xff03f000 Size=0x00200000
[0000.310] Non secure mode. Disable rollback prevention
[0000.315] AOTAG Init Done
[0000.352] Using GPT Primary to query partitions
[0000.358] Loading Tboot-CPU binary
[0000.407] Verifying bootloader in OdmNonSecureSBK mode
[0000.417] Bootloader load address is 0xa0000000, entry address is 0xa0000258
[0000.424] Bootloader downloaded successfully.
[0000.428] Downloaded Tboot-CPU binary to 0xa0000258
[0000.433] MAX77620_GPIO1 Configured.
[0000.436] MAX77620_GPIO5 Configured.
[0000.440] CPU power rail is up
[0000.442] CPU clock enabled
[0000.446] Performing RAM repair
[0000.449] Updating A64 Warmreset Address to 0xa00002e9
[0000.466] Bootloader DTB Load Address: 0x83000000
[0000.483] Kernel DTB Load Address: 0x83080000
[0000.488] Bootloader is not valid
[0000.491] Error in NvTbootLoadBinary: 0x14 !
[0000.495] GPT: Partition NOT found !
[0000.498] Find Partition via GPT Failed
[0000.502] Find Partition via PT Failed
[0000.506] function NvTbootGetBinaryOffsets: 0x1 error
[0000.510] Error in NvTbootLoadBinary: 0x1 !
[0000.514] Error is 1

And the following from pyro:

[0032.845] partition DTB write successful.
[0032.856] Writing BCT partition
[0032.860] bct_init bctinit
[0032.942] partition BCT write successful.
[0032.950] Change state to cold boot mode
[... 386 lines cut ...]
[0010.837] -----------------------------------------------
[0010.842] iframe 0x92c9a600:
[0010.845] x0  0x        92c64000 x1  0x               0 x2  0x             260 x3  0x               2
[0010.854] x4  0x               a x5  0x        92c9a6e0 x6  0x              20 x7  0x        6d365b1f
[0010.864] x8  0x               a x9  0x               7 x10 0x        92c9260d x11 0x             401
[0010.873] x12 0x        10000000 x13 0x              40 x14 0x               1 x15 0x        92cb6958
[0010.882] x16 0x           ffff0 x17 0x           10000 x18 0x               0 x19 0x        92c90000
[0010.892] x20 0x        92c64518 x21 0x        83028a30 x22 0x        92c64000 x23 0x        92c90000
[0010.901] x24 0x             400 x25 0x              24 x26 0x        92c9a9c0 x27 0x               0
[0010.910] x28 0x               0 x29 0x        92c9a920 lr  0x        92c38940 sp  0x        92c9a6f0
[0010.919] elr 0x        92c39c94
[0010.923] spsr 0x        60000309
[0010.926] -----------------------------------------------
[0010.931] panic (caller 0x92c01238): die
[0010.935] HALT: spinning forever...

2. I can workaround the above problem by changing the u-boot image created to the NVIDIA provided one in flash.xml (changing u-boot-jetson-tx1.bin to u-boot-dtb.bin).  However, when I flash this way, the DTB doesn't seem to be included in the main image.  I checked the ext3 file and it doesn't have the DTB in /boot and listing the contents of /boot on the eMMC in u-boot shows that it is missing there as well.

How can I resolve these issues? Both morty and pyro exhibit the same behaviors (though I really like not relying on the NVIDIA JetPack install for the flashing tools).  Is there a specific hash I need to be on with poky and meta-openembedded?

What am I doing wrong?

And on a related note I'd like to know the best way of specifying a different DTB for my hardware.  I'm utilizing carrier boards from ConnectTech and they provide DTB files defining their hardware.  How can I make those override the default DTB?

Thanks!

madisongh, 2017-12-13:
First, are you trying to use L4T R24.2 or R28.1?  The bootloaders, and the way U-Boot is handled, are completely different in the two releases, and you can't mix and match.  With R24, U-Boot is loaded directly by nvtboot and does not have the device tree bundled.  With R28, U-Boot is loaded by cboot and is packaged like an Android kernel image, with the device tree.

Did you make any changes to the U-Boot source?

Are you setting `IMAGE_FSTYPES = "tegraflash"` in your build configuration and using the `doflash` script in the resulting zip package to flash the device, or did you use some other method?

In meta-tegra, the morty and pyro branches are based off L4T R24.2.1 for the TX1.  The pyro-l4t-r28.1 branch uses R28.1.  You should be able to use the latest off the correspondingly-named branches in poky/oe-core and meta-openemebedded, although TBH I haven't tested morty lately, as I've been focused on pyro and rocko.

As for the custom device trees, if the vendor provided source files, you should be able to customize the kernel build to include them (along with any other kernel patches the vendor might have provided).  You might want to use a different MACHINE name and set KERNEL_DEVICETREE in the machine configuration file to point to the correct dtb file.

jhaws1982, 2017-12-13:
Shop,

Sent from my Verizon, Samsung Galaxy smartphone

jhaws1982, 2017-12-13:
I just double checked what version of JetPack I had installed and it is version 3.0, which should be R24.2.1 for the TX1 - so I believe that should match.  I built with branch pyro-l4t-r24.2.2 (after cleaning tmp and cache, my sstate and downloads are elsewhere).

Flashed from my JetPack 3.0 and it boots just fine now.  I also noticed that it is generating an ext4 instead of ext3, which it wasn't doing before and the DTB was included in that image.

I must have had some weird cross-configuration problem that was messing me up.  

Thanks for the help!

closed 2017-12-13 by jhaws1982

madisongh, 2017-12-13:
Good to hear that that cleared up the problem.   And yes, you should be able to use the existing jetson-tx1.conf as a starting point, then customize as needed.