OE4T/meta-tegra #59: getting error oe_runmake failed for tx2

madhumithavenkat, 2017-12-26:
Hi,
i am trying to build an image for tx2. I have changed the machine name to jetson-tx2 and added IMAGE_CLASSES += "image_types_tegra" in the local.conf file
I am getting the following error when i execute the command bitbake core-image-sato

ERROR: linux-tegra-4.4+gitAUTOINC+79e4600e81-r0 do_compile: oe_runmake failed
ERROR: linux-tegra-4.4+gitAUTOINC+79e4600e81-r0 do_compile: Function failed: do_compile (log file is located at /home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/temp/log.do_compile.10904)
ERROR: Logfile of failure stored in: /home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/temp/log.do_compile.10904

madisongh, 2017-12-26:
Which branch are you working with?  What other layers are included in your build, and what versions/branches of those?  What errors were reported in the `log.do_compile` file mentioned?

For L4T R28.1-based builds, I'd recommend using the `pyro-l4t-r28.1` branch here along with `pyro` in OE-core.  That should be the most stable combination.

madhumithavenkat, 2017-12-26:
i have cloned rocko branch and have only included meta-tegra layer. meta-tegra is also cloned from the rocko branch.

The log.do_compile file has the following
DEBUG: Executing shell function do_compile
NOTE: make -j 4 HOSTCC=gcc  -isystem/home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/recipe-sysroot-native/usr/lib -L/home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/recipe-sysroot-native/lib -Wl,-rpath-link,/home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link,/home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/recipe-sysroot-native/lib -Wl,-rpath,/home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/recipe-sysroot-native/usr/lib -Wl,-rpath,/home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/recipe-sysroot-native/lib -Wl,-O1 HOSTCPP=gcc  -E LIBGCC= Image CC=aarch64-poky-linux-gcc   -fuse-ld=bfd -ffile-prefix-map=/home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/git=/kernel-source/  LD=aarch64-poky-linux-ld.bfd  
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  GEN     ./Makefile
  CHK     include/generated/utsrelease.h
  Using /home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source as source for kernel
make[3]: 'include/generated/mach-types.h' is up to date.
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    /home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CC      arch/arm64/kernel/topology.o
  CC      arch/arm64/kernel/sys32.o
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c: In function 'parse_cluster':
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c:114:40: error: '%d' directive output may be truncated writing between 1 and 10 bytes into a region of size 3 [-Werror=format-truncation=]
   snprintf(name, sizeof(name), "cluster%d", i);
                                        ^~
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c:114:32: note: directive argument in the range [0, 2147483647]
   snprintf(name, sizeof(name), "cluster%d", i);
                                ^~~~~~~~~~~
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c:114:3: note: 'snprintf' output between 9 and 18 bytes into a destination of size 10
   snprintf(name, sizeof(name), "cluster%d", i);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c:129:37: error: '%d' directive output may be truncated writing between 1 and 10 bytes into a region of size 6 [-Werror=format-truncation=]
   snprintf(name, sizeof(name), "core%d", i);
                                     ^~
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c:129:32: note: directive argument in the range [0, 2147483647]
   snprintf(name, sizeof(name), "core%d", i);
                                ^~~~~~~~
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c:129:3: note: 'snprintf' output between 6 and 15 bytes into a destination of size 10
   snprintf(name, sizeof(name), "core%d", i);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c:59:39: error: '%d' directive output may be truncated writing between 1 and 10 bytes into a region of size 4 [-Werror=format-truncation=]
   snprintf(name, sizeof(name), "thread%d", i);
                                       ^~
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c:59:32: note: directive argument in the range [0, 2147483647]
   snprintf(name, sizeof(name), "thread%d", i);
                                ^~~~~~~~~~
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/arch/arm64/kernel/topology.c:59:3: note: 'snprintf' output between 8 and 17 bytes into a destination of size 10
   snprintf(name, sizeof(name), "thread%d", i);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/scripts/Makefile.build:261: recipe for target 'arch/arm64/kernel/topology.o' failed
make[3]: *** [arch/arm64/kernel/topology.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/home/madhumitha/yocto/tx2/tmp/work-shared/jetson-tx2/kernel-source/Makefile:973: recipe for target 'arch/arm64/kernel' failed
make[2]: *** [arch/arm64/kernel] Error 2
make[2]: *** Waiting for unfinished jobs....
  CHK     kernel/config_data.h
Makefile:150: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
Makefile:24: recipe for target '__sub-make' failed
make: *** [__sub-make] Error 2
ERROR: oe_runmake failed
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile (log file is located at /home/madhumitha/yocto/tx2/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/temp/log.do_compile.4464)

madisongh, 2017-12-26:
Try setting `GCCVERSION = "6.%"` in your `local.conf` if you're using the `rocko` branch.  The default compiler for `rocko` is GCC 7, and the NVIDIA kernel needs some additional patches to build properly with that version of the compiler.

Note, though, that even GCC 6 will not be fully compatible... specifically, the version of CUDA in L4T R28.1 only supports GCC 5 and earlier.  So if you will be adding any packages to your build that need CUDA, you need to pull in recipes for GCC 5 and configure your build to use that version of the compiler.

When L4T R28.2 is released, GCC 6 should be fully compatible - the R28.2 developer preview kit included CUDA 9.x, which supports GCC 6.

madhumithavenkat, 2017-12-29:
hi,
Thanks for the reply! I tried to build image using the pyro branch.
But i get the following error
Multiple versions of tegra-libraries are due to be built (/home/appsbuild/Linux/source/tx2/pyro/poky/meta-tegra/recipes-bsp/tegra-binaries/tegra-libraries_28.1.0.bb /home/appsbuild/Linux/source/tx2/pyro/poky/meta-tegra/recipes-bsp/tegra-binaries/tegra-libraries_27.1.0.bb). Only one version of a given PN should be built in any given build. You likely need to set PREFERRED_VERSION_tegra-libraries to select the correct version or don't depend on multiple versions.

madisongh, 2017-12-29:
For TX2 builds with OE-core pyro, use the `pyro-l4t-r28.1` branch.

madhumithavenkat, 2018-01-02:
Hi,
For pyro, i cloned yocto project pyro branch 2.3.3 and meta-tegra pyro-l4t-r28.1 branch. I have also cloned meta-openembedded of pyro branch.

When i try to build an image, i get the following error

ERROR: core-image-sato-1.0-r0 do_rootfs: Could not invoke dnf. Command '/home/appsbuild/madhumitha/pyro-tx2/tmp/work/jetson_tx2-poky-linux/core-image-sato/1.0-r0/recipe-sysroot-native/usr/bin/dnf -y -c /home/appsbuild/madhumitha/pyro-tx2/tmp/work/jetson_tx2-poky-linux/core-image-sato/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/home/appsbuild/madhumitha/pyro-tx2/tmp/work/jetson_tx2-poky-linux/core-image-sato/1.0-r0/rootfs/etc/yum.repos.d --repofrompath=oe-repo,/home/appsbuild/madhumitha/pyro-tx2/tmp/work/jetson_tx2-poky-linux/core-image-sato/1.0-r0/oe-rootfs-repo --installroot=/home/appsbuild/madhumitha/pyro-tx2/tmp/work/jetson_tx2-poky-linux/core-image-sato/1.0-r0/rootfs --setopt=logdir=/home/appsbuild/madhumitha/pyro-tx2/tmp/work/jetson_tx2-poky-linux/core-image-sato/1.0-r0/temp --nogpgcheck install dnf rpm run-postinsts packagegroup-core-x11-base packagegroup-core-x11-sato-games packagegroup-core-boot gstreamer1.0-omx-tegra packagegroup-core-x11-sato psplash packagegroup-base-extended packagegroup-core-ssh-dropbear locale-base-en-us locale-base-en-gb' returned 1:
Added oe-repo repo from file:///home/appsbuild/madhumitha/pyro-tx2/tmp/work/jetson_tx2-poky-linux/core-image-sato/1.0-r0/oe-rootfs-repo
Last metadata expiration check: 0:00:00 ago on Tue Jan 02 04:38:15 2018 UTC.
Error: 
 Problem 1: conflicting requests
  - nothing provides tegra-configs-omx-tegra needed by gstreamer1.0-omx-tegra-1.0.0+r28.1.0-r0.aarch64_tegra186
 Problem 2: package packagegroup-core-x11-sato-games-1.0-r33.jetson_tx2 requires puzzles, but none of the providers can be installed
  - conflicting requests
  - nothing provides libgdk-pixbuf-2.0-0 >= 2.36.5 needed by puzzles-1:0.0+git0+8dfe5cec31-r0.aarch64
 Problem 3: package dnf-2.0.0+git0+f0093d672d-r0.aarch64 requires librepo0, but none of the providers can be installed
  - package librepo0-1.7.20+git0+e1137cbbda-r0.aarch64 requires libcurl4 >= 7.53.1, but none of the providers can be installed
  - conflicting requests
  - nothing provides libgnutls30 >= 3.5.9 needed by libcurl4-7.53.1-r0.aarch64
 Problem 4: package packagegroup-core-x11-base-1.0-r1.noarch requires matchbox-terminal, but none of the providers can be installed
  - package matchbox-terminal-0.1-r0.aarch64 requires libvte-2.91-0 >= 0.46.1, but none of the providers can be installed
  - conflicting requests
  - nothing provides libgdk-pixbuf-2.0-0 >= 2.36.5 needed by libvte-2.91-0-0.46.1-r0.aarch64
 Problem 5: package packagegroup-core-x11-sato-1.0-r33.jetson_tx2 requires packagegroup-core-x11-sato-apps, but none of the providers can be installed
  - package packagegroup-core-x11-sato-apps-1.0-r33.jetson_tx2 requires sato-screenshot, but none of the providers can be installed
  - conflicting requests
  - nothing provides libgdk-pixbuf-2.0-0 >= 2.36.5 needed by sato-screenshot-0.3-r0.aarch64
 Problem 6: package packagegroup-base-extended-1.0-r83.jetson_tx2 requires packagegroup-base, but none of the providers can be installed
  - package packagegroup-base-1.0-r83.jetson_tx2 requires packagegroup-machine-base, but none of the providers can be installed
  - conflicting requests
  - nothing provides tegra-configs-nvstartup needed by packagegroup-machine-base-1.0-r83.jetson_tx2
(try to add '--allowerasing' to command line to replace conflicting packages)

ERROR: core-image-sato-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/appsbuild/madhumitha/pyro-tx2/tmp/work/jetson_tx2-poky-linux/core-image-sato/1.0-r0/temp/log.do_rootfs.2431
ERROR: Task (/home/appsbuild/Linux/source/tx2/pyro/poky/openembedded-core/meta/recipes-sato/images/core-image-sato.bb:do_rootfs) failed with exit code '1'

madisongh, 2018-01-03:
I have pushed an update to the pyro-l4t-r28.1 branch to remove the L4T 27 recipes to prevent older versions getting selected (although I hadn't run into that before - at least not in the same way).  That conflict may have led to some of the other conflicts that you are seeing, so give it a try with the latest.

There may be something else going on that's unrelated to this, though...

madhumithavenkat, 2018-01-12:
Hi,
With the latest commit the mentioned errors did not occur! Thank you :)

madhumithavenkat, 2018-01-15:
Hi, 
I had to include ostree in my build. I was able to build it for tx1. for tx2, i am facing an issue.
ERROR: ostree-1.0-r0 do_configure: Function failed: do_configure (log file is located at /home/appsbuild/madhumitha/new_tx2/tmp/work/aarch64-poky-linux/ostree/1.0-r0/temp/log.do_configure.30377)
ERROR: Logfile of failure stored in: /home/appsbuild/madhumitha/new_tx2/tmp/work/aarch64-poky-linux/ostree/1.0-r0/temp/log.do_configure.30377
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-64', 'arm-common', 'arm-64', 'common-linux', 'common-glibc', 'aarch64-linux', 'common']
| DEBUG: Executing shell function autotools_preconfigure
| Previously configured separate build directory detected, cleaning /home/appsbuild/madhumitha/new_tx2/tmp/work/aarch64-poky-linux/ostree/1.0-r0/build
| DEBUG: Shell function autotools_preconfigure finished
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-64', 'arm-common', 'arm-64', 'common-linux', 'common-glibc', 'aarch64-linux', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_configure
| You don't have gtk-doc installed, and thus won't be able to generate the documentation.
| autoreconf: Entering directory `.'
| autoreconf: configure.ac: not using Gettext
| autoreconf: running: aclocal --force -I buildutil -I libglnx ${ACLOCAL_FLAGS}
| configure.ac:54: warning: macro 'AM_PATH_GLIB_2_0' not found in library
| configure.ac:121: warning: macro 'AM_PATH_GPGME_PTHREAD' not found in library
| autoreconf: configure.ac: tracing
| autoreconf: running: libtoolize --copy --force
| libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
| libtoolize: copying file 'build-aux/ltmain.sh'
| libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'buildutil'.
| libtoolize: copying file 'buildutil/libtool.m4'
| libtoolize: copying file 'buildutil/ltoptions.m4'
| libtoolize: copying file 'buildutil/ltsugar.m4'
| libtoolize: copying file 'buildutil/ltversion.m4'
| libtoolize: copying file 'buildutil/lt~obsolete.m4'
| configure.ac:54: warning: macro 'AM_PATH_GLIB_2_0' not found in library
| configure.ac:121: warning: macro 'AM_PATH_GPGME_PTHREAD' not found in library
| autoreconf: running: /home/appsbuild/madhumitha/new_tx2/tmp/work/aarch64-poky-linux/ostree/1.0-r0/recipe-sysroot-native/usr/bin/autoconf --force
| configure.ac:54: error: possibly undefined macro: AM_PATH_GLIB_2_0
|       If this token and others are legitimate, please use m4_pattern_allow.
|       See the Autoconf documentation.
| autoreconf: /home/appsbuild/madhumitha/new_tx2/tmp/work/aarch64-poky-linux/ostree/1.0-r0/recipe-sysroot-native/usr/bin/autoconf failed with exit status: 1
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/appsbuild/madhumitha/new_tx2/tmp/work/aarch64-poky-linux/ostree/1.0-r0/temp/log.do_configure.30377)

madisongh, 2018-01-15:
Looks like the recipe may be missing a couple of dependencies, based on the warnings being emitted.  It's unlikely that the problem is specific to building for the TX2.

madhumithavenkat, 2018-01-18:
Hi,
Sorry! Its nothing related to tx2. Found out where the problem was!
Thank you :)

madisongh, 2018-01-20:
Great.  Since the original build problem has been resolved, I'm going to go ahead and close this.

closed 2018-01-20 by madisongh