OE4T/meta-tegra #61: do_image_complete fails: missing file: tmp/deploy/licenses/u-boot-tegra/recipeinfo

mwcoop, 2018-01-26:
Hi,

I am new to Yocto but have been bringing myself up-to-speed. I am attempting both core-image-minimal and core-image-sato builds for the TX2. My setup steps:

- Checkout pyro branch of Yocto
- Checkout pyro-l4t-r28.1 branch of meta-tegra
- oe-init-build-env as usual
- Added the meta-tegra layer to bblayers.conf
- Added the following to local.conf:
MACHINE ?= "jetson-tx2"
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
GCCVERSION = "5.%"  # Cuda support

The build seems to progress nearly to completion but fails in the core-image-sato recipe do_image_complete() function. It gives a python traceback showing that it is trying to write the license manifest. The exception thrown is:

FileNotFound: <path>/tmp/deploy/licenses/u-boot-tegra/recipeinfo

I am running this inside a preconfigured Docker image of Ubuntu 14.04 with all Yocto dependencies present. (I can build the basic default image for qemu/x86.) Any help is greatly appreciated.

madisongh, 2018-01-26:
Does this happen consistently, or only sometimes?  There may be a missing dependency that is leading to a race between populating u-boot's license info and its consumption by the do_image_complete task.

If you add this to your local.conf, does the problem go away?
```
do_image_tegraflash[depends] += "u-boot:do_populate_lic"
```

mwcoop, 2018-01-26:
The problem is very consistent on my systems. Of course, race conditions are a unique beast. Your suggestion does solve the issue. Oddly enough this caused an additional ~800 tasks to run compared to what was running before.

I am new to the scale of Yocto so find it daunting to deal with race conditions like this. :)

madisongh, 2018-01-26:
Yeah, races like this can be very dependent on specific hardware configurations on the build host.  I'm just surprised I haven't run into it myself before.

Glad that workaround helped... I'll do a bit more investigating and put a fix in place for this, when I get a chance.

mwcoop, 2018-01-26:
And it works for both minimal and sato which is good.

Not to hijack this completely but where are the files for flashing for TX2? Not seeing things that seem to match your Wiki page.

madisongh, 2018-01-26:
The zip package at `tmp/deploy/images/jetson-tx2/core-image-{minimal,sato}-jetson-tx2.tegraflash.zip` contains everything you should need.  Just unzip that somewhere and execute the `doflash.sh` script there (under sudo if required for access to the USB device).

mwcoop, 2018-01-27:
Got it.

If I can provide anything from my system to demonstrate the issue, please let me know.

Matt

commit d92a0300 (2018-01-27): image_types_tegra.bbclass: add missing dependencies

to ensure that license files are populated under tmp/deploy
for image license manifest creation.

Fixes #61.
commit 1e7a8e7f (2018-01-27): image_types_tegra.bbclass: add missing dependencies

to ensure that license files are populated under tmp/deploy
for image license manifest creation.

Fixes #61.
commit c19e9bc1 (2018-01-27): image_types_tegra.bbclass: add missing dependencies

to ensure that license files are populated under tmp/deploy
for image license manifest creation.

Fixes #61.
closed 2018-01-27 by madisongh