OE4T/meta-tegra #62: Failure in Kernel recipe - do_package

avishorp, 2018-02-01:
Building against Yocto Rocko (2.4), after commit 7c89171b6865de9171f54de2ec493220971b24e3, the Kernel recipe fails at the packaging stage:

```
ERROR: linux-tegra-4.4+gitAUTOINC+79e4600e81-r0 do_package: QA Issue: linux-tegra: Files/directories were installed but not shipped in any package:
  /boot/initrd
  /boot/extlinux
  /boot/extlinux/extlinux.conf
```

madisongh, 2018-02-01:
That commit went into master to account for some recent OE-core changes that went into master.
You should use the rocko branch here with the rocko branch in OE-core/poky.

shastri91, 2018-03-20:
Hey, 
Im new to yocto i am getting the same error. Can you please give me more knowledge on using rocko branch in OE-poky. Im not able to understand what that is.. please help.

ERROR: linux-tegra-4.4+gitAUTOINC+79e4600e81-r0 do_package: QA Issue: linux-tegra: Files/directories were installed but not shipped in any package:
  /boot/initrd
  /boot/extlinux
  /boot/extlinux/extlinux.conf
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
linux-tegra: 3 installed and not shipped files. [installed-vs-shipped]
ERROR: linux-tegra-4.4+gitAUTOINC+79e4600e81-r0 do_package: Fatal QA errors found, failing task.
ERROR: linux-tegra-4.4+gitAUTOINC+79e4600e81-r0 do_package: Function failed: do_package
ERROR: Logfile of failure stored in: /home/nithish/yocto/builds/sdr/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+79e4600e81-r0/temp/log.do_package.1564
ERROR: Task (/home/nithish/yocto/OE-core/poky/meta-tegra/recipes-kernel/linux/linux-tegra_4.4.bb:do_package) failed with exit code '1'

Thank You

madisongh, 2018-03-20:
The metadata layers are stored in git repositories, and the different versions are tracked on different branches in those repositories.   So if you cloned/checked out the 'rocko' branch in OE-core (or poky), you should also check out the branch in meta-tegra that has the same name.  The convention for most layers is to name their branches to correspond to the branch in openembedded-core/poky.

The 2.4 release from the Yocto Project was code named 'rocko', and that's the name of the branch where the tags for that release (2.4, 2.4.1, 2.4.2) have been placed.  I'm not sure the YP documentation makes the code name to version number mapping very clear.

Hope this helps.

shastri91, 2018-03-23:
Thank you !! worked without a glitch. 
But while I'm trying to flash i get this error,
Writing partition sc7 with warmboot_wbheader.bin.encrypt
[   4.3312 ] [................................................] 100%
[   4.3401 ] File to be written cannot be of zero size
[   4.3448 ] 
Error: Return value 10
Command tegradevflash_v2 --pt flash.xml.bin --create

can you please help me with this ..

[TX2_Console.log](https://github.com/madisongh/meta-tegra/files/1840216/TX2_Console.log)
[Host_Console.log](https://github.com/madisongh/meta-tegra/files/1840217/Host_Console.log)

madisongh, 2018-03-24:
I think that would be the writing of the u-boot image.  Does `u-boot-jetson-tx2.bin` exist in the directory where you unpacked the tegraflash zip file, and is it of non-zero length?

Here's what the subsequent output should look like:
```
[  12.3390 ] Writing partition sc7 with warmboot_wbheader.bin.encrypt
[  12.3396 ] [................................................] 100%
[  12.3478 ] Writing partition kernel with u-boot-jetson-tx2.bin
[  12.3526 ] [................................................] 100%
[  12.3730 ] Append 496 to tegra186-quill-p3310-1000-c03-00-base.dtb
[  12.3832 ] Writing partition kernel-dtb with tegra186-quill-p3310-1000-c03-00-base.dtb
[  12.3840 ] [................................................] 100%
[  12.4061 ] 
[  12.4081 ] tegradevflash_v2 --write BCT br_bct_BR.bct
[  12.4098 ] Bootloader version 01.00.0000
[  12.4118 ] Writing partition BCT with br_bct_BR.bct
[  12.4123 ] [................................................] 100%
[  12.4582 ] 
[  12.4649 ] tegradevflash_v2 --write MB1_BCT mb1_cold_boot_bct_MB1_sigheader.bct.encrypt
[  12.4667 ] Bootloader version 01.00.0000
[  12.4687 ] Writing partition MB1_BCT with mb1_cold_boot_bct_MB1_sigheader.bct.encrypt
[  12.4694 ] [................................................] 100%
[  12.5158 ] 
[  12.5158 ] Flashing completed

[  12.5159 ] Coldbooting the device
[  12.5178 ] tegradevflash_v2 --reboot coldboot
[  12.5196 ] Bootloader version 01.00.0000
```

shastri91, 2018-03-31:
Hey, the u-boot-jetson-tx2.bin was missing worked well after copying it. Thank you very much

closed 2018-04-07 by madisongh