OE4T/meta-tegra #64: TX2: Adding cuda-toolkit to core-image-minimal

niweis, 2018-03-07:
Hi,

I successfully built core-image-minimal and flashed it to the TX2. My Build Configurations:
  BB_VERSION            = "1.34.0"
  BUILD_SYS              = "x86_64-linux"
  NATIVELSBSTRING   = "universal-4.8"
  TARGET_SYS            = "aarch64-poky-linux"
  MACHINE                = "jetson-tx2"
  DISTRO                    = "poky"
  DISTRO_VERSION     = "2.3.3"
  TUNE_FEATURES       = "aarch64"
  TARGET_FPU             = ""
  meta              
  meta-poky         
  meta-yocto-bsp        = "pyro:e31d17d2e1a7a011f5ba0f14f392763846bd80fd"
  meta-tegra               = "pyro-l4t-r28.1:c19e9bc1f5b058fbae3ebd118e5379aee296fe86"

Added the meta-tegra layer to bblayers.conf and following to local.conf:
  MACHINE ?= "jetson-tx2"
  IMAGE_CLASSES += "image_types_tegra"
  IMAGE_FSTYPES = "tegraflash"
  GCCVERSION = "5.%" # Cuda support

Now I want to add the cuda toolkit to the core-image-minimal. Therefore I created a new layer:
  ./poky/scripts/yocto-layer create tx2-cuda
  mkdir -p ./meta-tx2-cuda/recipes-core/images
Added the following to new recipe meta-tx2-cuda/recipes-core/images/core-image-tx2-cuda.bb:
  DESCRIPTION = "Minimal image with cuda installation."
  LICENSE = "MIT"
  require recipes-core/images/core-image-minimal.bb
  IMAGE_INSTALL += "cuda-toolkit"

The exception thrown is:
ERROR: Task do_populate_sdk in /home/user/work/meta-tx2-cuda/recipes-core/images/core-image-tx2-cuda.bb rdepends upon non-existent task do_package_write_rpm in /home/user/work/meta-tegra/recipes-devtools/cuda/cuda-toolkit_8.0.84-1.bb
ERROR: Command execution failed: 1

Furthermore I was a little bit confused about the cuda-toolkit_8.0.84-1.bb "dummy recipe". Did it work in the past? I did the same procedure on the TX1 (morty branch). Everthing worked fine there. Any help is greatly appreciated.

madisongh, 2018-03-08:
The task is `do_populate_sdk` - were you trying to create an SDK?

I changed the cuda-toolkit recipe after morty so that the individual packages could be installed as needed (based on what other recipes link to), rather than the entire toolkit (which is really large).  That's when I changed cuda-toolkit to its current form.  It inherits 'nopackages' so no packages get created for it, so it can't be installed into an image (or SDK, I guess).

It probably makes sense to have that recipe be more like a 'meta' package that just RDEPENDS on the individual packages in the list, which would make it work more like it did in morty.  In the meantime, you could change your IMAGE_INSTALL to include the individual packages listed in the cuda-toolkit recipe, if you really want them all installed.

niweis, 2018-03-12:
I didn't try to create an SDK - don't know where it come from.

Tried to include the individual packages listed in the cuda-toolkit recipe. Everthing worked fine. I think you can close the issue.

Thanks for the help!

commit 46b11534 (2018-04-07): cuda-toolkit: provide runtime pseudo-package

for the CUDA 9 version of the recipe, so that the
individual packages don't have to be enumerated for
inclusion in the rootfs.

(fixes #64)
commit dad3cd9e (2018-04-13): cuda-toolkit: provide runtime pseudo-package

for the CUDA 9 version of the recipe, so that the
individual packages don't have to be enumerated for
inclusion in the rootfs.

(fixes #64)
closed 2018-04-07 by madisongh