OE4T/meta-tegra #74: How to occupy whole 32Gb of TX2 eMMC for tegraflash option.

ilyaikik, 2018-04-11:
seems that setting IMAGE_ROOTFS_SIZE doesn't help.
Thank you!

ilyaikik, 2018-04-11:
Successfull with IMAGE_ROOTFS_SIZE = "12582912", but NOT with IMAGE_ROOTFS_SIZE = "30021632".

Is it possible to use all 30021632KiB?

Or with 12582912 how to add one more partition?

madisongh, 2018-04-11:
If you're creating a tegraflash package, try setting `ROOTFSPART_SIZE = "30064771072"` in your `local.conf`.  That should be 28GiB, matching what the L4T flash configuration uses.

I'll update that setting the jetson-tx2 machine configuration.   I must have copied over the value from the jetson-tx1 configuration, forgetting that the TX2 has a larger MMC.

ilyaikik, 2018-04-11:
Nope.
That's not working. Creates partitionwith default size.
`Filesystem                Size      Used Available Use% Mounted on
none                      7.6G         0      7.6G   0% /dev
/dev/mmcblk0p1            1.4G      1.2G    145.2M  89% /
tmpfs                     7.7G    208.0K      7.7G   0% /run
tmpfs                     7.7G    196.0K      7.7G   0% /var/volatile
`

ilyaikik, 2018-04-13:
Nope.
When doing doflash.sh:
doflash.sh stalls on:
`[   5.7444 ] Writing partition APP with smp-image-xfce.img`
`[   5.7711 ] [                                                ] 000%`
while Jetson serial-port show:
`[0030.298] I> Writing APP partition`
`[0030.302] I> Total write size is 1261527788 bytes`
`[0030.346] E> 198cc004: Failed to write partition`

madisongh, 2018-04-13:
That's an improvement, it's just that the IMAGE_ROOTFS_SIZE is too large to fit into the partition.  Try `IMAGE_ROOTFS_SIZE = "29360128"`, which is 28 * 1024 * 1024 - the value is the size in 1KiB blocks.  I just tried that combination, and it worked OK for me.

madisongh, 2018-04-14:
I've pushed changes to master, sumo, and rocko-l4t-r28.2, so if you build off any of those branches, you should now be able to just set IMAGE_ROOTFS_SIZE in your local.conf to increase the size of the rootfs installed via doflash.   The 28*1024*1024 value I mentioned before is a good one to use for the TX2.

ilyaikik, 2018-04-16:
Sorry for late respond =)
I've tried this
`ROOTFSPART_SIZE = "30064771072"`
`IMAGE_ROOTFS_SIZE = "29360128"`
`IMAGE_ROOTFS_SIZE_pn-tegra186-minimal-initramfs = "8192"`
on pyro
and it work!
Thank you very much!

closed 2018-04-19 by ilyaikik