OE4T/meta-tegra #76: vulkan-demos don't appear to work in core-image-sato

sronsse, 2018-04-12:
Hello,

First of thank you for this meta and all the work you put in it!

I am currently testing out vulkan on the Jetson TX1 and demos appear to be working correctly using the NVIDIA Ubuntu release. I compiled these demos under Yocto (vulkan-demos recipe) under the jetson-tx1 environment (had to force them to use X and not Wayland by changing the recipe). The "triangle" binary is the one I used to validate the functionality (even though other demos work as well).

When building core-image-sato and using the flashing script generated using the "tegraflash" image FS type, one of the first issues I noticed was that the Vulkan ICD file was not present in the resulting rootfs. This description file points to the NVIDIA driver (libGLX_nvidia.so.0) and is automatically found by the vulkan loader (libvulkan.so.1) under /etc/vulkan/icd.d/ - it is normally copied by the "apply_binaries.sh" shell script in the NVIDIA JetPack package. Copying that file over allows "triangle" to start execution, but unfortunately fails when creating a surface through xcb with the following error:
/usr/src/debug/vulkan-demos/git-r0/git/base/VulkanSwapChain.hpp:253: void VulkanSwapChain::initSurface(xcb_connection_t*, xcb_window_t): Assertion `res == VK_SUCCESS' failed.

I am going to continue investigating this issue, but wanted to know if you had already tried this on your end or if there is something obvious I could be missing. Thanks!

sronsse, 2018-04-13:
Thanks - I'll dig into dependencies as well and see if I can find something.

sronsse, 2018-04-13:
Quick update on my end, which sounds both promising and scary at the same time :) I built a core-image-x11 image, and after manually adding the vulkan ICD file + copying all the libxcb* files from NVIDIA's /usr/lib/ into the resulting rootfs, I can now get vulkan to run (even though startx /usr/bin/triangle throws some exception in the first run - still works though). I'm now going down the path of figuring out which exact library did the trick.

sronsse, 2018-04-13:
Luckily, libxcb is fine as it is from the default build - the issue is the missing libxcb-glx.so* files. Copying only those from NVIDIA's rootfs does the trick. I don't think this is NVIDIA-specific, and believe that these are simply not built as part of libxcb in Yocto. I'll give you an update as I figure it out.

sronsse, 2018-04-13:
Alright, figured it out - libxcb under Yocto splits extensions into separate packages. Adding libxcb-glx fixes the issues I was seeing :) That means the only missing portion in the meta is the ICD file in order to get vulkan working!

madisongh, 2018-04-14:
I've updated the tegra-libraries recipe to install the ICD file, and have successfully built and run the vulkan-demos package out of OE-core in my core-image-sato image.  Thanks for tracking this down.

sronsse, 2018-04-14:
Great, thanks! Out of curiosity, did you have to modify the vulkan-demos recipe to force it to use xcb instead of wayland? I haven't tested the base recipe yet after fixing the other issues I ran into.

madisongh, 2018-04-15:
Yes, I added a bbappend to set `PACKAGECONFIG = "xcb"` so it would build for X rather than Wayland.

closed 2018-06-07 by madisongh