OE4T/meta-tegra #80: tegra-flashtools-native failed

BarbareBzhdev, 2018-04-18:
Hi @madisongh,

When I build our droneOS with Meta-tegra:rocko I have a little error : 
```
ERROR: tegra186-flashtools-native-28.1.0-r0 do_install: Function failed: do_install (log file is located at /build/tmp/work-shared/L4T-native-tegra186-28.1.0-r0/temp/log.do_install.145)

ERROR: Logfile of failure stored in: /build/tmp/work-shared/L4T-native-tegra186-28.1.0-r0/temp/log.do_install.145 
```

If I check L4T-native-tegra I don't find shell script, but I can found it in /build/tmp/work-shared/L4T-tegra186-28.1.0-r0 (no native version).
So for fix it I set Meta-tegra/recipes-bsp/tegra186-flashtools-native-28.1.0.bb with no native directory.

Is it normal? How can I fix it with better way?

Thanks :)

madisongh, 2018-04-18:
Ooops...I had fixed this for the 28.2 recipe, but forgot to make the same change for 28.1 at the time.  Try the latest, should work better.

BarbareBzhdev, 2018-04-18:
Nice, thank you very much :+1:

commit 4f5c977e (2018-04-18): tegra-binaries: make flash helper script common

Even though its tegra186-specific, the recipe is -native,
so the SoC-specific override being used to conditionally
bring in the script is in appropriate.

(Fixes #80)
closed 2018-04-22 by madisongh