OE4T/meta-tegra #81: alternative for libdrm

ElioDream, 2018-04-19:
Hey @madisongh (yeah, me again, sorry ^^)
I'm looking to install libdrm to my image, but every time it seems that it only installs libdrm-tegra... Is there a way to force the installation of the freedesktop libdrm ? Maybe add ALTERNATIVE_... to a bbappend for libdrm2.4.83 ?
I'll continue looking into it

madisongh, 2018-04-19:
I think if you set a higher ALTERNATIVE_PRIORITY in a bbappend for the normal libdrm, that would take care of it.

ElioDream, 2018-04-19:
That's what you did already, actually...
It really doesn't want to install. I removed the whole tmp and bitbaked
again

Damien KIRK
Élève ingénieur a Télécom Physique Strasbourg
06-64-25-24-70

ElioDream, 2018-04-19:
I see in ./recipes-bsp/tegra-binaries/tegra-libraries_28.1.0.bb:36:    rm -f ${D}${libdir}/libdrm* that all libdrm files are removed from the libdir. Could it be that the default libdrm tries to install before, and is removed when tegra-libraries is installed ? And by the way, do I have to specify for those packages to be included in my image (tegra-*) ? I'd say I don't, but I'll ask nonetheless ^^

madisongh, 2018-04-19:
Sorry, 'higher' priority is actually a lower ALTERNATIVE_PRIORITY value.  The libdrm-tegra recipe sets a priority of 10, where has the libdrm bbappend in the meta-tegra layer sets a priority of 20.  So libdrm-tegra wins by default.  If you want to override this, use a priority of < 10 in your libdrm bbappend.

Since it is a shared library, the build system will automatically install it if you include other packages that link against libdrm.

Note that the main reason for favoring the NVIDIA-provided copy is that NVIDIA doesn't support the underlying kernel DRI/DRM interface, so in order to get the X server (or anything else that uses the libdrm API) to work, NVIDIA's libdrm has to be there.

ElioDream, 2018-04-19:
That's why...
I should have thought about this earlier.
So what you say is that I HAVE to use the nvidia libdrm, or that it has to be present ?
In the first case, I guess I'll just write a pkg-config file for the libdrm-tegra...
Something surprising, though : I moved the libdrm2.83 package to my board via ftp, and installed it. I was able to launch a GUI...

madisongh, 2018-04-19:
Interesting... I'll need to go back and revisit this, then.  I originally omitted the NVIDIA pre-built copy, then ran into some issues that convinced me that it was needed, but I don't remember now what those were.  It would simplify things if we could exclude it.

So when you say it didn't change anything, does that mean `/usr/lib/libdrm.so.2.4.0` and `/usr/lib/libdrm.so.2` are still symlinking to `libdrm-tegra` instead of `libdrm-freedesktop`?  Hmm.

ElioDream, 2018-04-19:
It's worse than that, libdrm-freedesktop isn't even installed...
And now I tried to write a bbappend that would only add libdrm.pc to the pkgconfig directory, and it doesn't want to find the file during the install.

My patch is a very simple one : 

FILESEXTRAPATHS_append := "${THISDIR}/${PN}"

do_install_append() {
    install -d ${D}${libdir}/pkgconfig
    install -m 0644 ${WORKDIR}/libdrm.pc ${D}${libdir}/pkgconfig
}

SRC_URI += " \
    file://libdrm.pc \
"
but I run into this error : 
install: cannot stat '/home/elio/Workspace/yocto/jetson/build/tmp/work/aarch64_tegra210-poky-linux/libdrm-tegra/28.1.0-r0/libdrm.pc': No such file or directory

closed 2018-04-22 by madisongh

madisongh, 2018-04-22:
I've removed all the special handling to try to allow for both the NVIDIA pre-built copy of libdrm and the copy built from the libdrm recipe to be part of the build. It obviously wasn't working as intended, and doesn't really appear to be necessary.

If it turns out that some other NVIDIA-provided package really does require their pre-built libdrm over the standard copy, we'll need to figure out some other way to accommodate that.

commit 3ec41a88 (2018-04-22): meta: remove libdrm-related hacks

which weren't doing what was intended, and appear to be unnecessary.

* Remove the libdrm-tegra recipe
* Remove tegra-libraries dependency on libdrm-tegra
* Remove the libdrm bbappend
* Remove the mesa dependency on libdrm-stubs

(Fixes #81)
commit 3b0e0216 (2018-04-22): meta: remove libdrm-related hacks

which weren't doing what was intended, and appear to be unnecessary.

* Remove the libdrm-tegra recipe
* Remove tegra-libraries dependency on libdrm-tegra
* Remove the libdrm bbappend
* Remove the mesa dependency on libdrm-stubs

(Fixes #81)
commit 2737de3d (2018-04-22): meta: remove libdrm-related hacks

which weren't doing what was intended, and appear to be unnecessary.

* Remove the libdrm-tegra recipe
* Remove tegra-libraries dependency on libdrm-tegra
* Remove the libdrm bbappend
* Remove the mesa dependency on libdrm-stubs

(Fixes #81)
commit 1060f053 (2018-04-22): meta: remove libdrm-related hacks

which weren't doing what was intended, and appear to be unnecessary.

* Remove the libdrm-tegra recipe
* Remove tegra-libraries dependency on libdrm-tegra
* Remove the libdrm bbappend
* Remove the mesa dependency on libdrm-stubs

(Fixes #81)