OE4T/meta-tegra #82: opencv recipe fails with nothing provides libopencv-ts

lfdmn, 2018-04-23:
On branch pyro-l4t-r28.1 adding opencv to the image fails with "Nothing provides libopencv-ts"

Browsing the web points to the test library being built as static library instead of shared. But the patches for fixing this issue are already applied. At least I think so.
* 0001-Make-opencv-ts-create-share-library-intead-of-static.patch
* 0001-To-fix-errors-as-following.patch

For now I added -DBUILD_TESTS=OFF to EXTRA_OECMAKE to prevent building the test library but I don't know how to really fix the issue.

madisongh, 2018-04-24:
I've never run into this, but I don't usually try to include all of opencv in my images.  I just include applications that link against opencv libraries, and let the build system pull in whatever libraries that get linked into those apps.  Setting BUILD_TESTS to OFF is probably a good solution, unless you need the tests.

lfdmn, 2018-06-14:
Closing this issues. I disabled building the test module.

closed 2018-06-14 by lfdmn