OE4T/meta-tegra #83: What is the correct way to build a library using the cuda class

lfdmn, 2018-04-23:
Hi,

I'm trying to build arrayfire library which depends in CUDA.

What is the correct way to do this? Should it be enough to inherit the cuda class as below?

```
#####################################################################################
## Arrayfire library 
#####################################################################################

DESCRIPTION = "Arrayfire library"
SECTION = "libs"

SRC_URI = "git://github.com/arrayfire/arrayfire.git;protocol=https;branch=v3.5"
SRCREV = "0a675e87f42455e65cef7ea09ccd3200a3dd1db7"

LICENSE = "BSD"
LIC_FILES_CHKSUM = "file://COPYRIGHT.md;beginline=23;endline=23;md5=36c365f981e36a20caa006f389db83d5"

DEPENDS = "freeimage"

EXTRA_OECMAKE += "-DBUILD_CPU=OFF -DBUILD_CUDA=ON -DBUILD_OPENCL=OFF -DBUILD_EXAMPLES=OFF -DBUILD_TEST=OFF -DBUILD_GRAPHICS=OFF -DCMAKE_BUILD_TYPE=Release"

S = "${WORKDIR}/git"

inherit cmake cuda
```

This fails for me with this error:
```
| CMake Error at src/backend/cuda/CMakeLists.txt:316 (MESSAGE):
|   CMake CUDA Variable CUDA_CUDA_LIBRARY Not found.
| 
| 
| CUDA Driver Library (libcuda.so/libcuda.dylib/cuda.lib) cannot be found.
| CMake Error at src/backend/cuda/CMakeLists.txt:330 (MESSAGE):
|   Ending CMake configuration because of missing CUDA_CUDA_LIBRARY
```

libcuda.so itslocated in 
`./tmp/work/aarch64_tegra186-poky-linux/arrayfire/3.5.1-r0/recipe-sysroot/usr/local/cuda-8.0/lib/stubs/libcuda.so
`
Other recipes like opencv have the library in both /usr/lib/ and /usr/local/cuda-8.0/lib/stubs/
```
./tmp/work/aarch64_tegra186-poky-linux/opencv/3.2+gitAUTOINC+70bbf17b13-r0/recipe-sysroot/usr/local/cuda-8.0/lib/stubs/libcuda.so
./tmp/work/aarch64_tegra186-poky-linux/opencv/3.2+gitAUTOINC+70bbf17b13-r0/recipe-sysroot/usr/lib/libcuda.so
```

A usage example would be very helpful.

Thanks

madisongh, 2018-04-24:
It really depends on what's in the makefiles/cmake files/etc. for the package you're building, as there seems to be a lot of variation in how packages use CUDA.  Looks like this one is failing in the do_configure step. I usually start turning on verbose logging in cmake in the `run.do_configure` file (adding `--debug-output` and `--trace`, to start with), then manually running that script and examining the output to see what's going on.

lfdmn, 2018-05-17:
Thanks for the tip it helped. 

I traced down the issue being that FindCUDA.cmake from cmake package does not look for the libraries in the ...lib/stubs directory.

OpenCV fixed the problem by shipping a fixed FindCUDA with their sources. See https://github.com/opencv/opencv/tree/master/cmake.

I've made a cmake append in my layer and waiting for build to complete to test if it works.

This might be something you'd be interested to add to the meta-tegra layer.

Then inherit cuda in the recipe should be sufficient to have all cuda dependencies for bitbake to build.

madscientist42, 2019-04-03:
Useful information, really.  Coming late to this one, I'm finding similar problems so I'll be looking at the bbclass there and looking at what gets fixed.  The problem is that everything is rather framed in in pathing that is "wrong" for a lot of the tools- they presume the stock Linux distro placement- which would MOSTLY work with Yocto, and either SDK type for the same to produce good binaries for target, etc.  The way they're placed in the Jetpack is...odd...for the other stuff.   And CMake being somewhat simplistic in it's finding methods (for GOOD reason), it can't make the leap gracefully in many of the contexts.

madisongh, 2019-06-15:
OK, so I finally got round to updating `cuda.bbclass` so it should work (or at least work better) with cmake when using its CUDA language support (as opposed to the older FindCUDA method).

closed 2019-09-01 by madisongh