OE4T/meta-tegra #86: Is it possible to build linux-tegra outside of yocto?

tompreston, 2018-06-05:
I'm trying to build the [linux-tegra repo](https://github.com/madisongh/linux-tegra) branch `patches-l4t-r28.2` from source (not using `bitbake linux-tegra`) with the tegra18_defconfig (Jetson TX2) so that I can quickly develop and test. However I'm getting the following error message:
```
drivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu/common/linux/kmem.c:30:25: fatal error: gk20a/gk20a.h: No such file or directory
 #include "gk20a/gk20a.h"
```

This is because the make command isn't building the correct gcc command with the required nvgpu include from `nvidia/nvgpu/drivers/gpu/nvgpu/Makefile.nvgpu`:
```
-Idrivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu
```

That is (using `make V=1`), yocto generates:
```
  aarch64-poky-linux-gcc -Wp,-MD,drivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu/common/linux/.kmem.o.d  -nostdinc -isystem /mnt/yoctobuilds/tpreston/muos/build/tmp/work/jetson_tx2-poky-linux/linux-tegra/4.4+gitAUTOINC+cc8d223c40-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../lib/aarch64-poky-linux/gcc/aarch64-poky-linux/6.3.0/include
-I./arch/arm64/include
-Iarch/arm64/include/generated/uapi
-Iarch/arm64/include/generated 
-I./include
-Iinclude
-I./arch/arm64/include/uapi
-Iarch/arm64/include/generated/uapi
-I./include/uapi
-Iinclude/generated/uapi -include ./include/linux/kconfig.h 
-I./drivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu
-Idrivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu -D__KERNEL__ -mlittle-endian 
-I./arch/arm/mach-tegra/include
-Iarch/arm/mach-tegra/include -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mgeneral-regs-only -DCONFIG_AS_LSE=1 -fno-pic -mpc-relative-literal-loads -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -Wno-frame-address -O2 --param=allow-store-data-races=0 -Wframe-larger-than=2048 -fno-stack-protector -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-var-tracking-assignments -g  
-I./nvidia/t18x/include  
-I./nvidia/nvhost/include  
-I./nvidia/display/include  
-I./nvidia/nvmap/include  
-I./nvidia/nvgpu/include -fno-inline-functions-called-once -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time 
-I./drivers/gpu/nvgpu/include
-Idrivers/gpu/nvgpu/include 
-I./drivers/video/tegra/host
-Idrivers/video/tegra/host 
-I./drivers/devfreq
-Idrivers/devfreq  
-I./nvidia/nvgpu/include  
-I./nvidia/nvgpu/include/uapi  
-I./nvidia/nvgpu/drivers/gpu/nvgpu/include  
-I./nvgpu/drivers/gpu/nvgpu -Wno-multichar -Werror -Wno-error=cpp -Wextra -Wno-unused-parameter -Wno-missing-field-initializers  
-I./nvidia/nvgpu/include    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(kmem)"  -D"KBUILD_MODNAME=KBUILD_STR(nvgpu)" -c -o drivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu/common/linux/kmem.o ./drivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu/common/linux/kmem.c
```

and building locally I get:
```
aarch64-linux-gnu-gcc -Wp,-MD,drivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu/common/linux/.kmem.o.d  -nostdinc -isystem /usr/lib/gcc-cross/aarch64-linux-gnu/6/include
-I./arch/arm64/include
-Iarch/arm64/include/generated/uapi
-Iarch/arm64/include/generated 
-Iinclude
-I./arch/arm64/include/uapi
-Iarch/arm64/include/generated/uapi
-I./include/uapi
-Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -mlittle-endian
-Iarch/arm/mach-tegra/include -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -mgeneral-regs-only -DCONFIG_AS_LSE=1 -fno-pic -mpc-relative-literal-loads -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -Wno-frame-address -O2 --param=allow-store-data-races=0 -Wframe-larger-than=2048 -fno-stack-protector -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-var-tracking-assignments -g
-I./nvidia/t18x/include
-I./nvidia/nvhost/include
-I./nvidia/display/include
-I./nvidia/nvmap/include
-I./nvidia/nvgpu/include -fno-inline-functions-called-once -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time
-Idrivers/gpu/nvgpu/include
-Idrivers/video/tegra/host
-Idrivers/devfreq
-I./nvidia/nvgpu/include
-I./nvidia/nvgpu/include/uapi
-I./nvidia/nvgpu/drivers/gpu/nvgpu/include
-I./nvgpu/drivers/gpu/nvgpu -Wno-multichar -Werror -Wno-error=cpp -Wextra -Wno-unused-parameter -Wno-missing-field-initializers
-I./nvidia/nvgpu/include    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(kmem)"  -D"KBUILD_MODNAME=KBUILD_STR(nvgpu)" -c -o drivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu/common/linux/kmem.o drivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu/common/linux/kmem.c
```

See the missing:
```
-I./drivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu
-Idrivers/gpu/../../nvidia/nvgpu/drivers/gpu/nvgpu
```
before
```
-D__KERNEL__ -mlittle-endian 
```

Is it possible to build linux-tegra outside of yocto? Am I missing some environment variables?

madisongh, 2018-06-05:
I don't know whether `tegra18_defconfig` is really usable for builds, and I suspect that's the problem.  Try using [this one from meta-tegra](https://github.com/madisongh/meta-tegra/blob/master/recipes-kernel/linux/linux-tegra-4.4/tegra186/defconfig), or another one that is derived from the L4T package. You can extract the `.config` file from the top-level directory of the `kernel_headers.tbz2` tarball and modify as needed.

tompreston, 2018-06-06:
`tegra18_defconfig` is for Jetson TX2 according to the L4T docs. Just to be clear, I'm under the impression that this repo a single-repo version of the `Linux_for_Tegra/sources` layout gathered from `Linux_for_Tegra/source_sync.sh` - for the purposes of building with yocto. These are the sources in Linux_for_Tegra:
```
$ ls Linux_for_Tegra/sources/
hardware  kernel  u-boot

$ ls Linux_for_Tegra/sources/kernel/
display  kernel-4.4  nvgpu  nvhost  nvmap  nvmap-t18x  t18x

$ ls Linux_for_Tegra/sources/hardware/nvidia/platform/*
Linux_for_Tegra/sources/hardware/nvidia/platform/t18x:
common  quill

Linux_for_Tegra/sources/hardware/nvidia/platform/t210:
common  jetson

Linux_for_Tegra/sources/hardware/nvidia/platform/tegra:
common
```

I still get the error when compiling the config from `meta-tegra`:
```
$ git branch
patches-l4t-r28.2
$ curl https://raw.githubusercontent.com/madisongh/meta-tegra/master/recipes-kernel/linux/linux-tegra-4.4/tegra186/defconfig  > .config
$ export CROSS_COMPILE=aarch64-linux-gnu-
$ export CROSS_COMPILE=aarch64-linux-gnu-
$ make oldconfig
```

However, I've just noticed that [this path](https://github.com/madisongh/linux-tegra/blob/patches-l4t-r28.2/nvidia/nvgpu/drivers/gpu/nvgpu/Makefile.nvgpu#L9) is incorrect. I will submit a pull-request.

tompreston, 2018-06-06:
The pull request is here: https://github.com/madisongh/linux-tegra/pull/1

madisongh, 2018-06-06:
Thanks for tracking that down.  I'm not sure why that issue doesn't happen in the context of a bitbake build.

tompreston, 2018-06-06:
I've got no idea either.

closed 2018-06-07 by madisongh