OE4T/meta-tegra #89: Cannot bbappend tegra-tools

lfdmn, 2018-06-14:
Hi,

I've made a tegra-tools_28.2.0.bbappend to change the default PM_CONFIG DEFAULT from 2 to 3 to get max performance.

```
FILESEXTRAPATHS_prepend := "${THISDIR}/files/tegra186:"
SRC_URI_prepend_tegra186 = "file://nvpmodel.conf "

do_install_append_tegra186() {
    install -d ${D}${sysconfdir}
    install -m 0755 ${B}/usr/sbin/nvpmodel ${D}${sbindir}/
    install -m 0644 ${WORKDIR}/nvpmodel.conf ${D}${sysconfdir}/nvpmodel.conf
    install -d ${D}${sysconfdir}/init.d
    install -m 0644 ${S}/nvpmodel.init ${D}${sysconfdir}/init.d/nvpmodel
    install -d ${D}${systemd_system_unitdir}
    install -m 0644 ${S}/nvpmodel.service ${D}${systemd_system_unitdir}
}
```
Would you have any idea why the nvpmodel.conf prepend is ignored and the file never ends up in ${WORKDIR}. 

I've made sure the paths are correct. nvpmodel.conf exists and if I put a typo like nvpmodel.conf_blabla bitbake throws a warning that it cannot find the file. So I'm sure the file is found.

I'm having this issue with this one single recipe only, none of the others in my build system so I'm a bit puzzled.

madisongh, 2018-06-14:
Yeah, if you want to replace the default configuration file it's a bit trickier than that, because the do_fetch step is actually performed by the tegra-binaries recipe.  So you would create a tegra-binaries_28.2.0.bbappend with something like this:
```
FILESEXTRAPATHS_prepend := "${THISDIR}/files"
SRC_URI += "file://nvpmodel.conf"

do_preconfigure_append() {
    mkdir ${S}/config-replacements
    cp ${WORKDIR}/nvpmodel.conf ${S}/config-replacements/
}
```
then in tegra-nvpmodel_28.2.0.bbappend (or tegra-tools_28.2.0.bbappend if your copy of the layer predates the split-out of the nvpmodel files that I did back in April):
```
do_install_append_tegra186() {
    rm ${D}${sysconfdir}/nvpmodel.conf
    install -m 0644 ${S}/config-replacements/nvpmodel.conf ${D}${sysconfdir}/
}
```
If the `rm` command fails, you've got an older copy of the layer where the conf file got installed in the wrong location (in the subdirectory `${sysconfdir}/nvpmodel/`), so either update or change the command to `rm -rf ${D}${sysconfdir}/nvpmodel*`.  I fixed the install location a couple of weeks ago.

Another approach to making this change would be to just add a `sed` command to modify the default configuration, rather than providing a replacement copy of the config file.  You would only need one bbappend file, and it would do something like this:
```
do_install_append_tegra186() {
    sed -i -e's,PM_CONFIG_DEFAULT=3,PM_CONFIG_DEFAULT=2,' ${D}${sysconfdir}/nvpmodel.conf
}
```

Or you could just add a separate recipe of your own with a systemd service or initscript that invoke the `nvpmodel` command to set the power model to the desired setting. You'd have to be sure that it executes after `nvpmodel.service` starts, but otherwise, it might be simpler to implement.

lfdmn, 2018-06-15:
Ok thanks a lot Matt, I got it solved. I learned new tricks today =)

closed 2018-06-15 by lfdmn