OE4T/meta-tegra #90: Error fetching Tegra210_Linux_R28.2.1_aarch64.tbz2

sronsse, 2018-06-20:
Hi @madisongh,

I have a nightly build active for TX1/TX2 which is now failing due to the following:
```ERROR: tegra210-flashtools-native-28.2.1-r0 do_fetch: Fetcher failure for URL: 'https://developer.download.nvidia.com/devzone/devcenter/mobile/jetpack_l4t/3.2.1/m8u2ki/JetPackL4T_321_b23/Tegra210_Linux_R28.2.1_aarch64.tbz2'. Unable to fetch URL from any source.```
Might be as simple as a link change in the recipe and probably due to the last R28.2.1 update. I haven't tried to look for a working link yet, so if you have that solved already let me know! Thanks.

madisongh, 2018-06-21:
Oops, that slipped through the cracks, and somehow didn't get caught in my own testing.  Thanks!

closed 2018-06-21 by madisongh

sronsse, 2018-06-21:
Thanks Matt!

commit d8aa1094 (2018-06-21): tegra210-flashtools-native: remove 28.2.1 recipe

(Fixes #90)
commit 76407f9f (2018-06-21): tegra210-flashtools-native: remove 28.2.1 recipe

(Fixes #90)
commit fe3b2cc7 (2018-07-11): tegra210-flashtools-native: remove 28.2.1 recipe

(Fixes #90)