OE4T/meta-tegra #91: Fetcher failure: cuda-binaries

aaronshawcs, 2018-07-10:
I am having the following problem:
ERROR: cuda-binaries-9.0.252-1-r0 do_fetch: Fetcher failure for URL: 'http://developer.download.nvidia.com/devzone/devcenter/mobile/jetpack_l4t/3.2GA/m892ki/JetPackL4T_32_b196/cuda-repo-l4t-9-0-local_9.0.252-1_arm64.deb'. Unable to fetch URL from any source.

Looking at some previous issues, it might be a wrong link? I'm trying to hunt for a working link, but I'm having some trouble knowing where to look. Any help would be greatly appreciated!

madisongh, 2018-07-11:
Yeah, if you're using the master branch here, I haven't yet pulled in the recipe changes for the updated SRC_URIs since NVIDIA release L4T 28.2.1.

Use either the `rocko-l4t-r28.2` branch or the `sumo` branch for now.

aaronshawcs, 2018-07-13:
Thank you, this is working now!

closed 2018-07-13 by aaronshawcs