OE4T/meta-tegra #92: NOTE: preferred version 6.% of gcc-______ not available (for item _____________)

aaronshawcs, 2018-07-13:
I have added GCCVERSION ?= "6.%" to local.conf, but when I go to build, I'm getting this:

`NOTE: preferred version 6.% of gcc not available (for item cpp-symlinks)
NOTE: versions of gcc available: 7.3.0
NOTE: preferred version 6.% of gcc not available (for item gcc-symlinks)
NOTE: versions of gcc available: 7.3.0
NOTE: preferred version 6.% of gcc-sanitizers not available (for item gcc-sanitizers)
NOTE: versions of gcc-sanitizers available: 7.3.0
NOTE: preferred version 6.% of gcc not available (for item g++)
NOTE: versions of gcc available: 7.3.0`
and so on.

Which inevitably results in:

'| /home/ashaw/poky/build_jetson/tmp/work/aarch64_tegra186-poky-linux/opencv/3.3+gitAUTOINC+87c27a074d_2a9d1b22ed_a62e20676a_34e4206aef_fccf7cd6a4-r0/recipe-sysroot/usr/local/cuda-9.0/include/crt/host_config.h:119:2: error: #error -- unsupported GNU version! gcc versions later than 6 are not supported!'

This is my 9th day of using Yocto, so maybe I'm missing something?

Thanks,
Aaron

madisongh, 2018-07-13:
Which `openembedded-core` or `poky` branch are you basing your builds off of?  If it's the master branch there, be aware that they have dropped support for gcc 6 for the upcoming YP 2.6/thud release.  Stick with either YP 2.5/sumo or 2.4/rocko, and use the corresponding branch in meta-tegra as mentioned [here](https://github.com/madisongh/meta-tegra/wiki/Which-branch-should-I-use%3F).

aaronshawcs, 2018-07-16:
So I was on Sumo, and it looks like gcc 6.4 [hasn't been supported since December.](https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?h=sumo&id=dbea61d086973d08be0bca73159bcc3b1bec26a8)

I checked out Rocko, and got the error:

'ERROR: Layer openembedded-layer is not compatible with the core layer which only supports these series: rocko (layer is compatible with sumo)'

The way I see it, I have two options:

a. add back these removed gcc 6.4 files.
b. build an image using the parent of this commit.

Is there an easier solution I'm missing?

madisongh, 2018-07-16:
That message is about the meta-openembedded/meta-oe layer.  Checkout the rocko branch in meta-openembedded as well, and you should be all set.

aaronshawcs, 2018-07-16:
Yup, this seems to have worked. Thank you!

closed 2018-07-16 by aaronshawcs