OE4T/meta-tegra #94: Jetpack Ubuntu without Jetpack?

aaronshawcs, 2018-07-28:
Is it possible to use yocto to flash the ubuntu image from Jetpack onto the tegra? 

If so, what steps do I need to take to do this?

Also if this is beyond the scope of meta-tegra, let me know so I can ask elsewhere.

madisongh, 2018-07-28:
Yes, that's not really what this is here for - it's intended for folks that want to build their own Linux distros.
If you're just looking for a way to flash your Jetson/Tegra board without having to use the JetPack GUI, check the Linux For Tegra documentation (which is included in a JetPack installation).

aaronshawcs, 2018-07-28:
Understandable. It appears Yocto is too modular for this project but Jetpack isn't modular enough. 

Currently, we have an arduous method of flashing with Jetpack to get the exact image we want which involves adding and removing lots of features. The ideal outcome is finding a way to replicate that image with Yocto to speed up the flashing process. 

That being said, it doesn't have to be an exact copy, it just needs to run everything correctly, while also having enough quality-of life features that folks can comfortably develop on it. 

So I suppose my real question is about how to add quality-of-life features to my image, which I can ask elsewhere.

closed 2018-07-28 by aaronshawcs