OE4T/meta-tegra #95: cudaGetDeviceCount returned 35

gjohnsontx, 2018-08-03:
Hello All,

I have been working on creating a custom image for my Jetson TX2.  I have been able to compile my own CUDA code and the samples.  I am not sure what I am missing in my image but when I run the deviceQuery sample I just get a cudaGetDeviceCount return 35 error.  I have seen this issue on x86 when the nvidia-uvm and nvidia-uvm-tools devices don't exist but the Jetson doesn't seem to have that.  I am installing the following CUDA pacakges into my image:
`cuda-samples
cuda-cudart
cuda-command-line-tools
cuda-core
cuda-cufft
cuda-cublas
cuda-core
cuda-curand
cuda-driver
cuda-nvrtc
`

Thanks for any help.

madisongh, 2018-08-03:
From searching around, that error indicates that there is a version mismatch between the CUDA driver and the CUDA libraries.  Which meta-tegra branch are you using?

gjohnsontx, 2018-08-03:
So I needed to update to cuda 9.0 but the cuda nvcc from 9.0 only supports gcc 6.4.  So I am using the rocko meta-tegra layer but then I overrode the cuda recipes with the 9.0 recipes in mainline.  Is the CUDA driver compiled into the kernel or is it separate?

Thanks

madisongh, 2018-08-03:
It's `libcuda.so.1.1`, installed by the tegra-libraries recipe.  There may be dependencies between that library and the kernel (or other parts of the BSP).

If you want CUDA 9, I'd suggest using the `rocko-l4t-r28.2` branch.  That will get you the L4T 28.2.1 BSP plus CUDA 9 from JetPack 3.2.

gjohnsontx, 2018-08-08:
That worked!  Thanks @madisongh for the help.  I didn't realize there was a relationship between L4T and the JetPack.  Thanks again.

closed 2018-08-08 by gjohnsontx