OE4T/meta-tegra #102: rocko-l4t-r28.2: broke the drm

BassemMohsen, 2018-08-30:
I tested the latest rocko-l4t-r28.2 and I can see that gl applications that need access to drm results an error of: `[FATL | default] Could not open DRM (NV) device`
I wonder what have changed from `rocko` branch to result that.

madisongh, 2018-08-30:
Can you provide a little more context?

BassemMohsen, 2018-08-31:
Sure, so I'm trying to test https://doc-snapshots.qt.io/qt5-dev/qtopengl-hellogles3-example.html which uses EGLFS for managing native windowing system in the platform
you can read about it here: http://doc.qt.io/qt-5/embedded-linux.html
I think it fails to launch with the error couldn't open DRM device.

I can see that user space libdrm is on target. It might be an issue with the kernel module for DRM then.
How do you verify or test the graphics in tegra is working correctly usually? Do you have simple gl application you test with? or how did you test this?

madisongh, 2018-08-31:
I don't actually do much graphics testing myself beyond X11 support (in the core-image-sato image). It looks like the Qt folks do, though. Are you using their 'boot2qt' setup as documented in their [TX1 and TX2 setup guide](https://doc.qt.io/QtForDeviceCreation/qtee-preparing-hardware-jetsontx1.html)? Looks like their build is based off the meta-tegra rocko-l4t-r28.2 branch.

madisongh, 2018-09-03:
OK, I think I know what changed - I backed out all of the hacks I had in place for trying to have the open-source libdrm used during the build while still installing the NVIDIA binary-only copy of libdrm for use at runtime (see #81). So that might be causing the difference you're seeing in behavior.

I've spent some time this weekend trying to come up with a more usable solution (well, hack, really) that restores the NVIDIA trimmed-down libdrm library for runtime use but prevents the problems we were seeing earlier where programs that need to run under QEMU during the build (for gobject introspection, etc.) would segfault because of the NVIDIA-specific libraries being brought in.

The changes are on the [wip-drm-102](https://github.com/madisongh/meta-tegra/tree/wip-drm-102) branch, if you want to take a look. It's based off sumo rather than rocko-l4t-r28.2, but once I've done some further testing, I'll push them to all current branches.

I've also added mesa demos for GLX and GLES2 to my test builds for more graphics testing.

madisongh, 2018-09-09:
I've got the changes queued up for pulling into sumo now, just waiting for some test builds to finish. Once they're in sumo, I'll look at back-porting them to rocko-l4t-r28.2 and forward-porting into master.

BassemMohsen, 2018-09-10:
Oh, nice work.
I use meta-qt5 for compiling gfx on target.
meta-boot2qt is not that mandatory for gfx, unless you plan to use QtApplicationManager for complex wayland compositing setup.

I see the changes in the WIP. I will take time this week to test them on target.
I understand the problem you are trying to solve with coexisting the opensource mesa drivers and nvidia drivers and the compatibility with the DRI.
Nice touch with the mesa-demos as well. :)

madisongh, 2018-09-11:
The changes are in both sumo and rocko-l4t-r28.2 now.  If you run into any issues, let me know.

PR #103 (2018-09-10): Restore libdrm, add Wayland and Weston support
See [this wiki page](https://github.com/madisongh/meta-tegra/wiki/Wayland-Weston-support-on-TX1-TX2) for more info.

Fixes #102 and #41.
closed 2018-09-11 by madisongh