OE4T/meta-tegra #104: rocko-l4t-r28.2 core-image-sato configure

keguo1119, 2018-09-17:
Hi, I am new about yocto and this project. I fellow the descriptions about the readme, wiki, and issues, however i can't build the core-image-sato  successfully(core-image-mininal is ok). Here is what I have down.
Get the openembedded-core from` https://github.com/openembedded/openembedded-core.git ` and the branch to rocko. Set the meta-tegra to rocko-l4t-r28.2. And directory structure tree is below
`bitbake  build  LICENSE  meta  meta-selftest  meta-skeleton  meta-tegra  oe-init-build-env  README  README.LSB  README.qemu  scripts`. (The bitbake is a copy from `git://git.yoctoproject.org/poky`.)
`~: cat .templateconf 
# Template settings
TEMPLATECONF=${TEMPLATECONF:-meta-tegra/conf}`
`source oe-init-build-env `
And edit the conf files manual，
`# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
	/home/xunyi/Documents/openembedded-core/meta \
	/home/xunyi/Documents/openembedded-core/meta-selftest \
	/home/xunyi/Documents/openembedded-core/meta-skeleton \
	/home/xunyi/Documents/openembedded-core/meta-tegra \
	"
`

`cat conf/local.conf 
IMAGE_CLASSES += "image_types_tegra"               
IMAGE_FSTYPES = "tegraflash"
MACHINE = "jetson-tx2"`

Then I try to use `bitbake  core-image-sato` to build the image but get some errors!
`bitbake core-image-sato
Loading cache: 100% |##################################################################################################################################################| Time: 0:00:00
Loaded 1403 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'virtual/libgles2' (but /home/xunyi/Documents/openembedded-core/meta-tegra/recipes-multimedia/gstreamer/gstreamer1.0-plugins-nveglgles_1.2.3-r28.2.1.bb DEPENDS on or otherwise requires it)
mesa PROVIDES virtual/libgles2 but was skipped: one of '['opengl', 'vulkan']' needs to be in DISTRO_FEATURES
NOTE: Runtime target 'gstreamer1.0-omx-tegra' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['gstreamer1.0-omx-tegra', 'gstreamer1.0-plugins-nveglgles', 'virtual/libgles2']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'gstreamer1.0-omx-tegra', 'gstreamer1.0-plugins-nveglgles', 'virtual/libgles2']`

Sorry for my poor English.
Can you give me some suggestions?

madisongh, 2018-09-17:
This message:
```
mesa PROVIDES virtual/libgles2 but was skipped: one of '['opengl', 'vulkan']' needs to be in DISTRO_FEATURES
```
is the important one. Try adding
```
DISTRO_FEATURES_append = " opengl"
```
to your `local.conf` file, and see if that helps.

keguo1119, 2018-09-17:
You means we can build the core-image-sato  just based on the  openembedded-core and the meta-tegra projects ? Other meta projects is not necessary ?

madisongh, 2018-09-17:
Yes, you should be able to do that.  However, you might find it easier to start by using the poky repository and the meta-poky reference distro, using the documentation on the [Yocto Project website](https://www.yoctoproject.org).

keguo1119, 2018-09-17:
You are right, I am building the core-image-sato. Thank you for your help!

madisongh, 2018-09-18:
Look in your `local.conf` file for the `EXTRA_IMAGE_FEATURES` setting, which is commented out by default.  Either set it to `"debug-tweaks"` or to`"empty-root-password allow-empty-password"` and you'll be able to log in as root with no password.

closed 2018-10-06 by madisongh