OE4T/meta-tegra #106: Plans to support L4T 31.0.1

BassemMohsen, 2018-09-18:
Hello Matt

Do you have any plans/roadmaps to support the new release for L4T 31.0.1?
TBH, I'm more interested in some new kernel features to use AppArmor.

Also, If you want. I might be able to help some Pull Requests for that

madisongh, 2018-09-18:
I've started looking at 31.0.1 and will probably use it to start putting together the metadata for Xavier support (my Xavier is due to arrive today).  I probably won't pull any 31.0.1-based recipes into any of the main branches unless there is a long delay until L4T 32 comes out (which should support both Xavier and TX2, and will have the kernel sources in git, rather than just a tarball).

Even though 31 looks like it would actually support both Xavier and TX2, NVIDIA is only advertising support for Xavier with that version, so I'd be careful about trying to use it for anything else.

BassemMohsen, 2018-09-19:
> my Xavier is due to arrive today
well, Congratulations ;)

> Even though 31 looks like it would actually support both Xavier and TX2
I tried yesterday their pre-built rootfs from L4T 31 on the TX2 and worked nicely so far. so, I think 31 is already supporting TX2.

Anyway, Thanks for sharing your plans :+1:

mwcoop, 2019-01-10:
Does this build core-image-sato? My first attempts with image definitions from sumo is failing with many missing dependencies including wireless-tools, libxext, various X11 items, and several meta-tegra items. I imagine thud made a few changes that are messing me up. Any key points would be great.

madisongh, 2019-01-10:
Yes, both core-image-sato and core-image-weston.  I just tested builds against the latest OE-Core thud changes that went in a couple of days ago and found a couple of minor issues in meta-tegra (now fixed), but nothing major.

mwcoop, 2019-01-10:
Any chance you can provide sample bblayers and local config files?

madisongh, 2019-01-10:
My test distro setup is [here](https://github.com/madisongh/test-distro).  I haven't updated the `thud-l4t-r31.1.0` branch there for a couple of weeks, but you can also just use the `thud` branch and in your workspace:
* update the `meta-tegra` submodule to the WIP branch. 
* rename the bbappend for `tegra186-minimal-initramfs` under `meta-testdistro` to `tegra-minimal-initramfs` (and removing the obsolete bbappend for `tegra210-minimal-initramfs`)
* update the `tegra-mmapi-samples_31.1.0.bb` recipe to comment out the DEFAULT_PREFERENCE setting

Then just `source` the `setup-env` script at the top level to create a build directory and populate it with the necessary config files, update `conf/local.conf` to set MACHINE to what you want, and you should be able to build either of those images. Be warned that the builds (for TX1, TX2, or Xavier) could take a while because I add some packages that pull in `opencv`. You can tweak the bbappends I have for `core-image-weston` and `packagegroup-core-x11-sato` to trim some of those out (in particular, `tegra-mmapi-samples`) to cut down on build times.

madisongh, 2019-01-11:
I have just updated the `thud-l4t-r31.1.0` branch in the test distro to bring all the layers up to date.

mwcoop, 2019-02-03:
Looking at the test-distro project, I'm just curious what image types you are working with. In my particular case, I basically want to build a complete image with all Tegra tools, samples, development headers/docs, etc. What image types get me close to this. For example, do any images include the VisionWorks samples? It looks like core-image-weston may get close to what I want. Is that correct.