OE4T/meta-tegra #108: Selecting GCC version for build and image

mwcoop, 2018-09-19:
So, I worked with meta-tegra a while back and was having good success. I have come back and am trying to get a build going for TX2 on the Sumo branch.

Since my use case uses Cuda, I have placed the following in local.conf:

GCCVERSION = "6.%"

Unfortunately, I'm getting a Make error saying that a later GCC version is being used. Why would this be the case?

Also, I need to install GCC directly into the final image. Can you tell me how to select 6.% for this built-in GCC toolchain?

madisongh, 2018-09-20:
Do you have recipes for gcc 6 in your layers somewhere?  OE-Core dropped gcc 6 in sumo.

mwcoop, 2018-09-20:
Ahh. That makes sense. Looks like the error is specifically with OpenCV anyway. Oddly, if I go back to Rocko, OpenCV fails saying that only versions 5 and earlier are supported.

Is there a meta layer that has older recipes? Do you have a recommended setup?

mwcoop, 2018-09-20:
Possibly related. My OpenCV version is 3.3.  I see that meta-tegra makes modifications for version 3.4.3.  Should I specifically be getting that version?

madisongh, 2018-09-20:
meta-openembedded/meta-oe updated to opencv 3.4.3 in sumo, so I had to update the bbappend accordingly.  If you're still using 3.3, just copy over the bbappend and patches from the older copy in rocko and apply them locally (e.g. in your distro layer).

As for older gcc recipes, I've tried a couple of different approaches.  I've used the meta-linaro-toolchain layer, which required a bit of work since they provide the full suite of gcc/binutils/glibc, and I had to tweak some things to account for that since I wasn't using all of the meta-linaro layers.  Lately I've just been copying the gcc recipes I need into my distro layer from the prior OE-Core release.  That appears to work OK, and haven't had to add any bbappends for OE-Core recipes to compensate for the older compiler, at least for what I've been using.

Another approach would be to try some hacks to get CUDA 9 to build when using gcc 7.  Searching around I've run into some blog posts about it, and it looks like it's possible to do, but I've been more nervous about using that approach than just falling back to an older gcc.

mwcoop, 2018-09-20:
Thanks for all of the info. Oddly my git clone from GitHub.com/openembedded/meta-openembedded uses opencv_3.3.bb on the sumo branch.

madisongh, 2018-09-20:
Oh, oops, you're right, I misremembered.  meta-oe *master* has moved to 3.4.3, but sumo is still 3.3.  But the bbappend for the recipe in meta-tegra sumo branch is also for 3.3.   I only updated it in master.

madisongh, 2018-09-21:
Try setting `SDKGCCVERSION = "7.%"` so that the older compiler doesn't get used for building the host-side bits of the SDK.  I have that in my distro configuration too.

closed 2018-09-22 by mwcoop