OE4T/meta-tegra #110: WLAN0 on TX2: "Could not read SSID from driver"

mwcoop, 2018-09-22:
My distro/image is building, flashing, and booting nicely. Wired ethernet is functional. Unfortunately, WLAN0 is not working for me. While "iw dev wlan0 scan" reports all device capabilities, wpa_supplicant is failing to interact with the device.

Relevant installed recipes: crda, iw, wireless-tools, wpa-supplicant

/etc/network/interfaces for wlan0:

auto wlan0
iface wlan0 inet dhcp
    wpa-conf /etc/wpa_supplicant.conf

wpa_supplicant run at boot as:

$ wpa_supplicant -B -P /var/run/wpa_supplicant.wlan0.pid -i wlan0 -c /etc/wpa_supplicant.conf -D wext

Error message when trying a scan operation inside the wpa_cli environment is:

<4>Could not read SSID from driver

Has meta-tegra been tested for WiFi functionality? Any tricks to getting this working? Help greatly appreciated as always.

madisongh, 2018-09-22:
I do periodically test it, and last I checked it worked fine.  I don't use wi-fi regularly though, so I'll check it again in my setup.  That's a TX2, and you're using the sumo branch, correct?

mwcoop, 2018-09-22:
Yes. Thanks.

philipc, 2018-09-22:
I've used wifi on the sumo branch recently. I don't have a record of what I did, but probably you need to delete the `-D wext` argument, since the driver supports `nl80211`.

mwcoop, 2018-09-23:
Great. changing the driver to nl80211 did the trick. Thanks!

closed 2018-10-06 by madisongh