OE4T/meta-tegra #111: Add CAN modules and autoload

mwcoop, 2018-09-22:
Looking around on the forums seems to indicate that some extra steps may be necessary to enable CAN functionality in the kernel. I see that the mttcan0 and mttcan1 devices are in the device tree but nothing is shown with "ifconfig -a".

Are extra options/recipes required to add the needed modules to the kernel?

madisongh, 2018-09-22:
You need to coax the relevant kernel modules into your image, such as
```
CORE_IMAGE_EXTRA_INSTALL += "kernel-module-mttcan kernel-module-can"
```
If you need them to be loaded automatically at boot time, add `"can mttcan"` to  your `KERNEL_MODULE_AUTOLOAD` setting.

I don't know much about CAN, so I don't know if those are actually the right modules, but hopefully that will give you a lead.

mwcoop, 2018-09-22:
Awesome. That is seriously helpful.

closed 2018-10-06 by madisongh