OE4T/meta-tegra #113: Getting Started/Flashing Guidance

dwalkes, 2018-10-04:
Hi,
Thanks for sharing this project!

I'm just getting started with the TX2 so apologies for the newbie questions.  I have experience with Yocto but none with the TX2.

I wanted to understand the effort required to build my own Yocto based distribution instead of the Ubuntu based one installed with the Jetpack installer from Nvidia.  

I can pull in this layer to my yocto build, change the MACHINE in local.conf to
`
MACHINE ??= "jetson-tx2"
`
When I do this, I'm able to build some deployment images in build/tmp/deploy/images/jetson-tx2 including rootfs and .bin files with bitbake core-image-minimal.  However it's not clear to me how I could flash these to the board using the ./flash.sh or similar script from the Nvidia Jetpack installer.

I see some comments about tegraflash image types in #29 and a reference to
`
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
`
When I add this I see an additional tegraflash.zip file in the deploy directory, but it’s still not obvious how I can flash this on the part.

I'm guessing this is something which is obvious to folks who are more experienced with the platform and probably documented somewhere I haven't found yet.  Any pointers would be helpful.

Thanks!
Dan

madisongh, 2018-10-05:
Check out the [wiki page](https://github.com/madisongh/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit) on this topic, hopefully that will help.

dwalkes, 2018-10-05:
Doh!  Yep that's what I was looking for.  I'd suggest a link from the README page although it looks like I was the first one who didn't find it on my own.  Sorry about that.