OE4T/meta-tegra #116: Deepstream Sdk

Gotrek77, 2018-11-07:
Hi,
Thanks for your hard work!
Do you think to include support to install Deepstream Sdk?
https://developer.nvidia.com/deepstream-sdk

madisongh, 2018-11-09:
The Deepstream SDK is still in "early access" stage, so access is limited, and that doesn't make it very suitable for a recipe here.   If/when NVIDIA releases it and makes it available for anonymous download like they do with the JetPack components, it shouldn't be too hard to put together a recipe for it.

madscientist42, 2019-04-03:
Since I've been tasked by the client with getting the Deepstream SDK on a TX2 SDK/Target Image, I guess I'll be working on something for that as well...  X-D

klutchell, 2019-12-04:
Hey @madisongh, since we now have a mirror set up for the login-only L4T components, is there a chance we can bring in deepstream-sdk support?

madisongh, 2019-12-06:
Sure. I'm not necessarily going to have time to put together recipes for it myself any time soon, so if there's a volunteer willing to submit a PR...

fasm22, 2020-05-25:
Any update about deepstream support?

fasm22, 2020-05-25:
Hi @madisongh , any update about Deepstream support?

mmadrigal, 2020-06-01:
Hi @madisongh ,

[RidgeRun](https://www.ridgerun.com/) is working on adding support for DeepStream. Hopefully, we can merge it into your main branch soon for the sake of community :) 

We might be also contributing with some additional features for completeness. 

At this point, anybody can look at our wiki page which is intended to help the community setting up the meta-layer and all Jetson features:

https://developer.ridgerun.com/wiki/index.php?title=Yocto_Support_for_NVIDIA_Jetson_Platforms

Best Regards,
-Marco
www.ridgerun.com

madisongh, 2020-11-02:
See #425 ... there are still some issues to work out around Python support, but thanks to the RidgeRun folks for getting this going.

closed 2020-11-02 by madisongh

mmadrigal, 2020-11-02:
Hi @madisongh , thanks! we are working with NVIDIA on the best way to integrate the python bindings to the project.