OE4T/meta-tegra #117: Vulkan support on Xavier

sronsse, 2018-11-08:
Hello,

Thank you for your continued support on this meta, it's been excellent so far!

I recently tested the sumo wip branch which has Xavier support, and while all the vulkan demos run, the actual frames are completely black, except for the overlayed UI elements. I was wondering if you could test this on your end if/when you got a chance, as it's possible my configuration may be botched. I used the standard sumo vulkan 1.0.65.2 recipe for this purpose and my understanding is that the vulkan version that comes with JetPack is newer (and vulkan works), so that could potentially be an issue even though everything is supposed to be backwards compatible.

Cheers!

madisongh, 2018-11-09:
I wound up rebasing the Xavier work onto a thud-based branch (wip-thud-l4t-31.0.1), and OE-Core thud has Vulkan 1.1.73.  Could you give that a try and see if it works better?  I had to migrate because the NVIDIA gstreamer binary-only plugins are built for gstreamer 1.14.

madisongh, 2018-11-24:
I finally got around to trying this and am also running into some issues on both Xavier and TX2 with L4T 31.1.0.  In my case, Vulkan can't even initialize because it gets an error when trying to get surface properties.  TX2 works fine on sumo and thud, with L4T R28.2.1.

I even tried using recipes to build the Vulkan loader and headers for the latest 1.1.85.0 version, and I get the same result.  So there's some interaction between the loader and NVIDIA's implementation in libGLX_nvidia.so that's causing problems.  Not sure what that is yet, though.

madisongh, 2018-11-29:
I finally tracked this down to a missing library that libGLX_nvidia.so loads dynamically at runtime. I have put bbappends in place for the vulkan recipes, and now the demos work fine.

commit 3b78dfd8 (2018-11-29): vulkan, vulkan-demos: add bbappends

NVIDIA's Vulkan support is for X only, not Wayland, so
remove wayland configuration from the loader and demos.
Also, the Vulkan driver in libGLX_nvidia.so dlopen()s
the libxcb and libxcb-glx libraries, so make sure they
are in the runtime dependencies.

Fixes #117.
commit 51bffe94 (2018-12-07): vulkan, vulkan-demos: add bbappends

NVIDIA's Vulkan support is for X only, not Wayland, so
remove wayland configuration from the loader and demos.
Also, the Vulkan driver in libGLX_nvidia.so dlopen()s
the libxcb and libxcb-glx libraries, so make sure they
are in the runtime dependencies.

Fixes #117.
commit 05a6e3e3 (2018-12-22): vulkan, vulkan-demos: add bbappends

NVIDIA's Vulkan support is for X only, not Wayland, so
remove wayland configuration from the loader and demos.
Also, the Vulkan driver in libGLX_nvidia.so dlopen()s
the libxcb and libxcb-glx libraries, so make sure they
are in the runtime dependencies.

Fixes #117.
closed 2018-11-29 by madisongh