OE4T/meta-tegra #121: error parsing tegra-minimal-initramfs.bb

d-wang03, 2019-02-14:
Hello, when I try to build core-image-sato, the bitbake returns an error:

ERROR: /home/supervisor/osbuilder/bstos/meta-tegra/recipes-core/images/tegra-minimal-initramfs.bb: No IMAGE_CMD defined for IMAGE_FSTYPES entry 'cpio.gz.cboot.bup-payload' - possibly invalid type name or missing support class

Is there any way to fix it ?

Best Regards

d-wang03, 2019-02-14:
It's my fault for missing extra image settings. 
Sorry to bother.

closed 2019-02-14 by d-wang03

MartinBagge, 2019-05-01:
How did you resolve this issue?

kekiefer, 2019-06-10:
What was your fix for this? I have been working around by bringing in `image_types_tegra` into tegra-minimal-initramfs.bb (kekiefer/meta-tegra@c7796aebf) and this works for me, and seems like a reasonable thing to do, but I'm wondering why it wasn't done this way.

madisongh, 2019-06-10:
Just add
```
IMAGE_CLASSES += "image_types_tegra"
```
to your `local.conf`.