OE4T/meta-tegra #124: Help adding support for USB webcam/Gstreamer RTP/UDP?

mwcoop, 2019-03-19:
I've been trying to get my build to support standard USB webcams. Anyone know how to add this support when using meta-tegra? Also, does the Nvidia version of Gstreamer support any web streaming protocols like RTP/UDP? Any tricks to enable them?

madisongh, 2019-03-23:
Have you looked at NVIDIA's "L4T Accelerated Gstreamer User Guide"? That has example gstreamer pipelines for different types of camera inputs and should indicate which gstreamer plugin packages you need to include in your builds.  For RTP and UDP support, you just use the standard plugins; no NVIDIA-specific modifications are needed. RidgeRun has some example pipelines on their [wiki](https://developer.ridgerun.com/wiki/index.php?title=Gstreamer_pipelines_for_Jetson_TX1).

mwcoop, 2019-03-28:
Anything required to install extra plugins. I currently add "gstreamer" to the list of items for the image.

At a minimum, I cannot get v4l2src in gst-launch-1.0

madisongh, 2019-03-29:
All the plugins are packaged separately, although there are meta-packages for the good, bad, and ugly plugins that you could use to load them all in.  I use [this page](https://gstreamer.freedesktop.org/documentation/plugins.html) to help with figuring out which specific plugins to install - the package name to use is the combination of the module name and plugin name in the table on that page, so the v4l2src element is in gstreamer1.0-plugins-bad-video4linux2.

closed 2019-11-23 by madisongh